ghdl/ghdl #1927: pyGHDL module is not found even though it's installed (with pip)

Godhart, 2021-12-02:
**Description**
Python fails on 'import pyGHDL' or import of it's submodules.
pyGHDL is installed with pip.

**Expected behaviour**
Trying to run simple example from pyVHDLModel docs, which uses pyGHDL. Expecting to pass, but no.

**How to reproduce?**
Have tried on host with fresh arch-based linux (manjaro) installation and on docker ubuntu based image.

Can be easily reproduced on docker with following Dockerfile
```
FROM ubuntu:20.04

RUN apt-get update \
 && apt-get install -y \
        python3 pip \
        ghdl \
 && apt-get clean

RUN pip3 install pyVHDLModel pyGHDL

ENTRYPOINT ["python3"]

CMD ["-c", "import pyGHDL; print('works!')"]
```

Simply build image and run it, output is:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pyGHDL'
```

Expected output: `works!`

**Context**
- OS: linux (manjaro, ubuntu), 64-bit
- Origin:
  - [x] Package manager: `apt-get`, `pacman`, `pip`
  - [ ] Released binaries: 
  - [ ] Built from sources: 

**Additional context**
Trying to use pyVHDLModel, just started from example shown at https://pypi.org/project/pyVHDLModel/

umarcor, 2021-12-02:
Hi @Godhart,

Nice to see interest in pyVHDLModel and pyGHDL!

pyGHDL is not published to PyPI yet. Hence, `pip3 install pyGHDL` won't work.

The recommended approach (see http://ghdl.github.io/ghdl/quick_start/python/index.html) is to do `pip install git+https://github.com/ghdl/ghdl.git@$(ghdl version hash)`, in order to get the version of pyGHDL which matches exactly the version of GHDL installed on the system. See, for instance, https://github.com/umarcor/osvb/blob/main/.github/workflows/Doc.yml#L24-L34. That is because pyGHDL depends on libghdl, which is installed with GHDL; even though pyGHDL does not need the ghdl binary. When installing pyGHDL, pyVHDLModel will be installed as a dependency, so you don't need to install it explicitly.

Alternatively, you can execute `pip install .` in the root of this repository. If you do so, I strongly recommend that you checkout exactly the same commit that matches `ghdl version hash`.

Note that pyGHDL has a tight dependency on the version of pyVHDLModel. I recommend to uninstall pyVHDLModel and let pyGHDL install the dependency it needs. Installing different versions of pyVHDLModel might work, but that is only usable for development purposes.

Let us know if you need any further guidelines.

Godhart, 2021-12-02:
@umarcor Thanks for your guidance!
With Docker file from your work flow it works, will look further.

BTW. Had no luck yet with my arch (due to hash of prebuilt app is unknown, build from AUR fails) and ubuntu:20.04 too (there is still old 0.36 or so version). Going try to get it working on arch soon, but for now will stick to docker...

umarcor, 2021-12-02:
@Godhart can you let us know which specific package from Arch/AUR are you using?

Godhart, 2021-12-02:
https://aur.archlinux.org/packages/ghdl-llvm-git
https://aur.archlinux.org/packages/ghdl-mcode-git
both are `2.0.0dev.r6542.gbbdff59e4-1`

Fails on building `python-pymetaclasses` (sources check fails)

umarcor, 2021-12-07:
@marzoul @xiretza, can you help here? Does `ghdl version hash` or `ghdl version ref` return anything on your Arch builds?

Ref: https://github.com/ghdl/ghdl/commit/0747980c3e8bf101abee2aad730f03da1411d318

Xiretza, 2021-12-07:
`ghdl version hash` works fine for `ghdl-*-git`, but the repo packages (built from tarballs) do return `unknown` (as expected, there is no git repo involved).

> Fails on building `python-pymetaclasses` (sources check fails)

Please report such bugs to the respective AUR pages - now that I know about it, I just rechecked, and the tarball for pymetaclasses 1.3.1 did indeed change. I'm investigating.

umarcor, 2021-12-07:
@Paebbels, the issue with pymetaclasses might be related to changes in [pyTooling](https://github.com/pyTooling/)?

Xiretza, 2021-12-07:
Ah yes, @Paebbels moved the repository from their personal namespace to @pyTooling, which caused all the files in the tarball to be renamed. I've pushed a fix to the AUR package.

Paebbels, 2021-12-07:
@Xiretza Oh I didn't know that is effects AUR too. Shouldn't GitHub maintain redirects, so Git doesn't fail after rename / moving it.

Currently many of my individual packages like pyMetaClasses are summarized in a single pyTooling package under a new namespace at GitHub. We're currently updating dependencies and it will be updated in GHDL soon too.

When you write *tarball* do you refer to the packages Python code in tar.gz format created by `setup.py sdist`? So it's necessary to push them to PyPI too? (But I didn't change PyPI for these old packages.)

Xiretza, 2021-12-07:
I'm talking about the release tarballs - https://github.com/Paebbels/pyMetaClasses/archive/refs/tags/v1.3.1.tar.gz does redirect to the new repo, but the top level directory in the tarball is now named after the new repo, not the old.

closed 2022-01-15 by tgingold