ghdl/ghdl #1930: Support identifiers longer than 1023 characters

imphil, 2021-12-08:
Currently GHDL only supports identifiers (e.g. port names) up to 1023 characters long and (cleanly) terminates with a message like `test_long_log_msg/test.vhd:13:10: identifier is too long (> 1023)` otherwise. It would be nice to allow arbitrarily long identifiers (as long as allowed by the VHDL spec).

*(There's no pressing need to change anything. Of course, such long identifiers are unlikely in manually written code [unless you're using German words as identifiers], but are happening in auto-generated code. Filing this issue to have something to point to in cocotb where we disable a test for GHDL due to this limitation.)*

alpyen, 2023-07-17:
I've checked the LRM for 87, 93, 00 and 08 mainly for the identifier sections
and general limits of source files, did some tests with CAE software and opened a PR.

Here's what I found:

None of the standards above mention any length limits regarding identifiers or lines which would,
in my understanding, allow arbritrary lengths as long as the grammatical rules are satisfied.
These are listed in section: LRM 13.3 for Vhdl87, 93, 00, and 15.4 for Vhdl08.

I'm reading a book about VHDL-Synthesis and Simulation and it claims the same with the note that CAE software might only check the first 32 letters. Unfortunetaly it does not state which software does so.

So it got me wondering how it is being implemented by industry software, and I tested it with various standards (87 - 08):
- Quartus Prime Lite 19.1 (09/2019)
- ModelSim Intel FPGA Starter Edition 10.5b (10/2016)
- Xilinx Vivado 2022.2 (10/2022)

I created a basic entity, disabled all optimizations and routed two signals
with their names as long as almost one megabyte but differing in the last character.

All three do compile. Quartus/Vivado do Synthesize, Implement and Generate a Bitfile.

The only thing that hangs/crashes seems to be Quartus' RTL-Viewer and its EDA Netlist Writer.
The former is unable to handle signals longer than ~1K (which Vivado has no problem with) and
crashes with a log report stating that the signal is already routed (it could only compare up to the ~1K limit).
The latter remains stuck at 4% if the signal name is longer than ~2K.

In conclusion this means that the current limit of 1023 characters is not compliant with the standards
but some tools still have a softcap on it so one can either increase the limit to maybe 4K or fix it.
I've submitted a PR #2468 to fix this issue.

GHDL compiles and simulates the entity I described above and another project without any issue.
If somebody could run a few tests to make sure I didn't break anything, that would be much appreciated 😄

While I was trying to fix this issue I stumbled upong something weird:
The string length for "Identifier_Buffer" in package Chap10 (vhdl/translation/trans.adb) has a length of 4K
while the "Max_Name_Length" limit is set to 1K (vhdl/vhdl-scanner.ads). Was this an oversight or am I missing something?

Thanks in advance!

tgingold, 2023-10-02:
Unlimited length has been implemented.

closed 2023-10-02 by tgingold