ghdl/ghdl #1932: Compilation ARM fails

zipotron, 2021-12-14:
**Description**
Fails compile target ARMv7 using LLVM

**Expected behaviour**
"make" be sucess

**How to reproduce?**
In a Raspberry PI 2 or ARM Chromebook with ARMv7 (I am using an Asus Flip C100pa)

` ./configure --prefix=/usr/local --with-llvm-config`
`make`

And the output:

`gnatmake: "/home/zipotron/tools/ghdl/src/grt/grt-change_generics.adb" compilation error`

umarcor, 2021-12-14:
Please, provide a complete reproducer: which operating system are you using, how did you get the sources, how did you install the dependencies and which versions are you using.

zipotron, 2021-12-15:
Yes, sure. I am using Slackware, I got the last source from github, yesterday. Dependencies, well, I have all needed dependencies as long as the GHDL release 1.0.0 is compiling in that way perfectly, that is why I report this as a bug, actually regresion.

zipotron, 2021-12-15:
Just forgot to mention, I have the slackware up to date, the gcc-gnat 10.2 If I remember good I could compile the GHDL from github 4 or 5 mounth ago without problems, I could compile yesterday last Yosys and NextPNR without issues... I hope I could help you a bit for catch the problem!

tgingold, 2021-12-17:
we really need a more complete report.  In particular the error messages.

zipotron, 2021-12-20:
Hello @tgingold ! Well, there is not a descriptive error message, just that line, but I copy-paste longer frame of the log:

`/home/zipotron/tools/ghdl/./src/grt/vhpi_user.h:37:1: warning: "/*" within comment [-Wcomment]
   37 | /* $Revision: 1315 $
      |
/home/zipotron/tools/ghdl/./src/grt/vhpi_user.h:38:1: warning: "/*" within comment [-Wcomment]
   38 | /* $Date: 2008-07-13 10:11:53 +0930 (Sun, 13 Jul 2008) $
      |
/home/zipotron/tools/ghdl/./src/grt/vhpi_user.h:39:1: warning: "/*" within comment [-Wcomment]
   39 | /* --------------------------------------------------------------------
      |
gcc -c  -g -Wall -g -o grt-cdynload.o /home/zipotron/tools/ghdl/./src/grt/grt-cdynload.c
gcc -c  -g -Wall -g -o fstapi.o /home/zipotron/tools/ghdl/./src/grt/fst/fstapi.c -I/home/zipotron/tools/ghdl/./src/grt/fst
gcc -c  -g -Wall -g -o lz4.o /home/zipotron/tools/ghdl/./src/grt/fst/lz4.c
gcc -c  -g -Wall -g -o fastlz.o /home/zipotron/tools/ghdl/./src/grt/fst/fastlz.c
mkdir grt
echo "with Grt.Backtraces.Jit;" > grt-backtraces-impl.ads
echo "package Grt.Backtraces.Impl renames Grt.Backtraces.Jit;" >> grt-backtraces-impl.ads
cd grt && MSYS2_ARG_CONV_EXCL="-aI;-gnatec" gnatmake -c -aI/home/zipotron/tools/ghdl/./src/grt -aI.. \
  -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY ghdl_main -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X  -cargs
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/ghdl_main.adb
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt.ads
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-errors.adb
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-main.adb
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-options.adb
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-rtis_binding.ads
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-std_logic_1164.adb
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-types.ads
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-astdio.adb
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-astdio-vhdl.adb
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-hooks.adb
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-stdio.ads
gcc -c -I./ -I/home/zipotron/tools/ghdl/./src/grt -I.. -gnatec/home/zipotron/tools/ghdl/./src/grt/grt.adc -gnatdY -g -gnatwe -gnatwa -gnatwC -gnatf -gnatw.X -I- /home/zipotron/tools/ghdl/src/grt/grt-change_generics.adb
grt-change_generics.adb:56:37: warning: source alignment (1) < alignment of "Ghdl_Rtin_Type_Enum" (4)
grt-change_generics.adb:98:56: warning: source alignment (1) < alignment of "Ghdl_Rtin_Subtype_Scalar" (4)
grt-change_generics.adb:106:58: warning: source alignment (1) < alignment of "Ghdl_Rtin_Subtype_Scalar" (4)
grt-change_generics.adb:119:56: warning: source alignment (1) < alignment of "Ghdl_Rtin_Subtype_Scalar" (4)
grt-change_generics.adb:159:49: warning: source alignment (1) < alignment of "Ghdl_Rtin_Subtype_Scalar" (4)
grt-change_generics.adb:215:58: warning: source alignment (1) < alignment of "Ghdl_Rtin_Subtype_Scalar" (4)
grt-change_generics.adb:253:34: warning: source alignment (1) < alignment of "Ghdl_Rtin_Object" (4)
grt-change_generics.adb:265:54: warning: source alignment (1) < alignment of "Ghdl_Rtin_Subtype_Composite" (4)
grt-change_generics.adb:267:49: warning: source alignment (1) < alignment of "Ghdl_Rtin_Type_Array" (4)
grt-change_generics.adb:282:51: warning: source alignment (1) < alignment of "Ghdl_Rtin_Subtype_Scalar" (4)
grt-change_generics.adb:373:40: warning: source alignment (1) < alignment of "Ghdl_Rtin_Object" (4)
gnatmake: "/home/zipotron/tools/ghdl/src/grt/grt-change_generics.adb" compilation error
make: *** [src/grt/Makefile.inc:140: grt/run-bind.adb] Error 4
`

tgingold, 2021-12-20:
These are warnings.  Try to configure with --disable-werror

zipotron, 2021-12-20:
Works! Shall someone (me, you or someone else) add this to the GHDL documentation?

closed 2021-12-21 by zipotron