ghdl/ghdl #1933: Bug with signal spy

Aksl-B, 2021-12-17:
Hello, I tumbled upon this message while running a testbench including the following line : 
`assert priorities = << signal .arbiter_test.my_arbiter.devicePriorities : priorities_t >> report "Error on priorities input";`

Here is the bug report : 
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/Data/Supaéro/Substit/Cours/СБИС/ЛР/lr2/
Command line:
ghdl -e --std=08 arbiter_test
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x55db560f5ae8 0x55db562d987b 0x55db562d1786 0x55db562d91e0 0x55db562947a1 0x55db562a45b4 0x55db562a4763 0x55db562a7b7f 0x55db562acd72 0x55db5631ca6c 0x55db562824e9 0x55db5627f63c 0x55db5628d7f8 0x55db563213db 0x55db562604a4 0x55db56195cbb 0x55db56324d2c 0x55db55f26b93 0x7f87131e00b1 0x55db55f251fc 0xfffffffffffffffe
******************************************************************

Have a nice day with it ;)

tmeissner, 2021-12-17:
GHDL doesn't support external names yet. See  #548 for example.

Aksl-B, 2021-12-17:
Ok, thank you for your answer !
I'm still a beginner in VHDL and I'm wondering how I can check that my component well received the configuration data sent by the test bench without spying on this internal signal... Do you have a clue ?

Aksl-B, 2021-12-17:
Update : I used the technique described here
https://stackoverflow.com/questions/40286725/how-can-i-get-internal-signals-to-testbench-in-vhdl-97-and-isim/40313051#40313051

Quickly : I defined a global signal in which the component writes and from which the testbench reads, it's not as elegant but it seems to work

closed 2021-12-17 by Aksl-B

tmeissner, 2021-12-17:
Yes, that's a possible solution.