ghdl/ghdl #1939: Two process DFF with case statement is synthesized incorrectly

eugmes, 2022-01-02:
**Description**
I'm trying to use two-process design method (https://www.gaisler.com/doc/vhdl2proc.pdf) with ghdl but I'm getting incorrect synthesis results if I use case statements in combinational process. Simple design that should result in a single DFF is contains latches when synthesized by yosys and reportedly Vivado (https://github.com/YosysHQ/yosys/issues/3141#issuecomment-1003728117).

**Expected behaviour**
In the following example both architectures should result in a single DFF being synthesized.

**How to reproduce?**

```vhd :file: dff.vhd
library IEEE;
use IEEE.std_logic_1164.all;

entity dff is
    port (clk : in std_logic; d, load : in std_logic; q : out std_logic);
end entity dff;

architecture with_case of dff is
    signal r, rin : std_logic;
begin
    comb : process (d, load) is
        variable v : std_logic;
    begin
        v := r;
        case load is
            when '1' => v := d;
            when others => null;
        end case;
        rin <= v;

        q <= r;
    end process comb;

    regs : process (clk) is
    begin
        if rising_edge(clk) then
            r <= rin;
        end if;
    end process regs;
end architecture with_case;

architecture with_if of dff is
    signal r, rin : std_logic;
begin
    comb : process (d, load) is
        variable v : std_logic;
    begin
        v := r;
        if load = '1' then
            v := d;
        end if;
        rin <= v;

        q <= r;
    end process comb;

    regs : process (clk) is
    begin
        if rising_edge(clk) then
            r <= rin;
        end if;
    end process regs;
end architecture with_if;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 ent.vhd
ghdl synth --std=08 --out=verilog dff with_case > dff_with_case.v
ghdl synth --std=08 --out=verilog dff with_if > dff_with_if.v
yosys -p "read_verilog dff_with_case.v; synth; show;"
yosys -p "read_verilog dff_with_if.v; synth; show;"
```

**Context**

- OS: macOS
- Origin:
  - [X] Package manager: `2.0.0-dev (1.0.0.r950.g8d512a44) [Dunoon edition]`

**Additional context**

`with_if` architecture results in an expected one DFF:
[image]

Incorrect synthesis result for `with_case` architecture:
[image]

maehne, 2022-01-05:
@eugmes: In both of your architectures, the process `comb` is lacking the signal `r` in the sensitivity list. This may explain the unexpected synthesis results. The rule for combinatoric processes is to always list all read input signals/ports by the process. Alternatively, you may use the keyword `all` in the sensitivity list starting with VHDL-2008.

tmeissner, 2022-01-05:
Such sensitivity lists are usually ignored by synthesis tools. Their incompleteness is more a problem during simulation, as the behavior of simulation could differ from the intended (and synthesized) behavior.