ghdl/ghdl #1940: VHDL 2008 matching case does not handle don't cares correctly

eugmes, 2022-01-02:
**Description**
`case?` statement implementation does not seem to handle don't care values (`'-'`).
 
**Expected behaviour**
`-` should match any value when used inside matching case statement.

**How to reproduce?**

This example should print "ok", it prints "failed, i=1".

```vhd :file: test_mcase.vhdl
library IEEE;
use IEEE.std_logic_1164.all;

entity test_mcase is
end entity test_mcase;

architecture tb of test_mcase is
    signal i : std_logic_vector(0 downto 0) := "1";
begin
    process (i) is
    begin
        case? i is
            when "-" => report "ok" severity note;
            when others => report "failed, i=" & to_string(i) severity error;
        end case?;
    end process;
end architecture tb;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 test_mcase.vhd
ghdl --elab-run --std=08 test_mcase
```

**Context**

- OS: macOS
- Origin:
  - [x] Package manager: `GHDL 2.0.0-dev (1.0.0.r950.g8d512a44) [Dunoon edition]`

**Additional context**
This seem to indicate that `case?` is not supported: https://github.com/ghdl/ghdl/issues/717. In that case it would be good if ghdl could print an error message. That could save somebody a lot of time debugging strange issues.

eugmes, 2022-01-02:
Apparently `synth` prints a warning: `test_mcase.vhd:13:13:warning: choice with meta-value is ignored`, but there is no warning with `--elab-run` or when using `vunit`.

umarcor, 2023-03-07:
FTR, there is now a test for this feature in https://github.com/VHDL/Compliance-Tests/blob/main/vhdl_2008/tb_matching_case.vhd. Ref: VHDL/Compliance-Tests#19.

Hinara, 2024-06-10:
I think this example should not compile instead if failing silently ?
`It is an error if a choice does not represent a value of the subtype other than the scalar value '–' or an array value containing '–' as an element.` IEEE Std 1076-2008 page 165

tgingold, 2024-06-10:
Can you clarify which example are you referring to ?

Hinara, 2024-06-10:
The example from the issue especially
```vhdl
            when "-" => report "ok" severity note;
```

tgingold, 2024-06-10:
I don't think it should fail to compile as the choice represent a value of the subtype.
The subtype is `std_logic_vector(0 downto 0)`.

tgingold, 2024-06-11:
I agree this part of specification is poorly written.  In addition, using `case?` for `std_ulogic` is useless (if you have a `'-'` choice, you cannot have another choice).

But I understand this part as `'-'` is allowed even if not part of the subtype, like:

```vhdl
library IEEE;
use IEEE.std_logic_1164.all;

entity test_mcase is
    port (
      res_o: out std_ulogic
    );
end entity test_mcase;

architecture tb of test_mcase is
    signal i : std_ulogic range 'U' to '1' := '1';
begin
    process (i) is
    begin
        case? i is
            when '-' => res_o <= '1';
            when others => res_o <= '0';
        end case?;
    end process;
end architecture tb;
```

Normally, `'-'` wouldn't be allowed as it is not in `'U' to '1'` range.  This part allows it.

Hinara, 2024-06-11:
I think I was especially confused by the fact that this  rule applies only to std_ulogic and forgot about the constraint part.
Make sense, thank you :) and sorry for bothering you ^^'