ghdl/ghdl #1945: Synth crash: double alias access inside a process

kammoh, 2022-01-06:
**Description**
A clear and concise description of what the issue is about.

**Expected behaviour**
Synth crashes when referencing an alias of another alias inside a process. Simulation works fine. Single aliasing works fine. Continuous assignment also seems to be fine. The same behavior with or without VHDL 2008 syntax and switch.
Side note: The usefulness of having multiple levels of aliasing is probably not immediately clear in this contrived example, but comes very handy in a large code-base.

**How to reproduce?**

```vhd :file: mwe.vhd
library ieee;
use ieee.STD_LOGIC_1164.all;

entity MWE is
    port(
        i : in  std_logic_vector(7 downto 0);
        o : out std_logic
    );
end entity;

architecture rtl of MWE is
    alias i_alias : std_logic_vector(7 downto 0) is i;
    -- this crashes:
    alias lower   : std_logic_vector(3 downto 0) is i_alias(3 downto 0);
    -- this works fine:
    -- alias lower : std_logic_vector(3 downto 0) is i(3 downto 0);
begin
    -- this works fine too:
    -- o <= '1' when lower(3 downto 0) = "0000" else '0';

    process(all)
    begin
        if lower = "0000" then
            o <= '1';
        else
            o <= '0';
        end if;
    end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl --synth --std=08 mwe.vhd -e MWE
```

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: macOS
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r907.g51d120d8) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/kamyar/src/lwc/LWC/hardware/dummy_lwc/
Command line:
ghdl --synth --std=08 ../LWC_rtl/mwe.vhd -e MWE
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:877
Load address: 0x1005df000
Call stack traceback locations:
0x1009eb831 0x100719496 0x1007216fc 0x1008ed474 0x1008ea6e3 0x1008e5f0b 0x1008e743d 0x1008c567c 0x1008ce5de 0x1008cef14 0x1008d0763 0x1008d095f 0x1008f8651 0x1008f8742 0x1008f1b5d 0x100923419 0x1008456a6 0x1009d463b 0x1005e4ef0
******************************************************************
```

**Additional context**

Xiretza, 2022-01-06:
Hmm, looks like a possible regression of #945?

kammoh, 2022-01-06:
It seems like a related issue, but maybe not a regression. In that example, the crash happens even without read/write of the alias. In the current case, crash happens _only_ when the alias is read (and maybe written?) inside a process.

Xiretza, 2022-01-06:
Yeah, I guess the testcase from back then really wasn't broad enough. The reproducer you posted is very finnicky indeed though, the crash requires a lot of things to be just right!

It actually is a regression by the way, 1.0.0 handles the reproducer fine.

commit 82bc0568 (2022-01-12): synth: handle alias of alias.  Fix #1945
closed 2022-01-12 by tgingold