ghdl/ghdl #1946: Assertion failure loading architecture with rising/falling edge distinction

AsumFace, 2022-01-06:
**Description**
Attempting to load the example below in yosys with the GHDL plugin results in the assertion failure shown below. Simulation of the larger design which the example is derived from works without any issues. 

Breaking the branches up into separate if ... end if; statements resolves the assertion failure.

**Expected behaviour**
Yosys should reach the end of the script without failure.

**How to reproduce?**
Heavily reduced example:

```vhd :file: circuit.vhd
library ieee;
use ieee.std_logic_1164.all;

entity circuit is
    port (
        clk : std_logic
    );
end entity;

architecture rtl of circuit is
    signal sig : std_logic;
begin
    process (clk)
    begin
        if falling_edge(clk) then
            -- assertion failure only as long as this branch exists
        elsif rising_edge(clk) then
            -- assertion failure only when assignment is here
            sig <= '1';
        end if;
    end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
yosys -m ghdl -p "ghdl --std=08 circuit.vhd -e circuit"
```

**Context**

- OS: Gentoo Linux
- Origin:
  - [x] Package manager: `Nix unstable, GHDL 1.0.0, yosys-0.12+36, yosys-ghdl-2021.01.25`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (tarball) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-unknown-linux-gnu
/top_secret_path/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-memories.adb:2575
******************************************************************
```

**Additional context**

tgingold, 2022-01-07:
Ok, it shouldn't crash but I am not sure if both edge processes should be handled.

dalex78, 2026-08-19:
# Issue #1946 — Crash: `falling_edge`/`rising_edge` branches in one process

Analysis by an AI agent. Recommend (optionally merging the test and) closing the ticket.

## High-level summary

A clocked process with `if falling_edge(clk) then ... elsif rising_edge(clk) then <signal assignment> end if;` (both edge checks in one `if`/`elsif` chain, rather than separate `if` statements) crashed `--synth` (originally hit via the Yosys GHDL plugin, but the crash is in GHDL's own synth backend, reproducible directly with `ghdl --synth`) with an internal `ASSERT_FAILURE` at `netlists-memories.adb:2575`.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report (with an output port added so the assigned signal is observable, avoiding dead-code elimination) against current `master`: `ghdl --synth --std=08 circuit.vhd -e circuit` now produces a complete netlist with a `gate_mdff` (mixed dual-edge D flip-flop) instantiation instead of crashing. This must have been fixed as part of general improvements to mixed-edge-detection handling in `src/synth` since this was reported (originally against GHDL 1.0.0, 2021). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1946/` (the repro from the report, with an output port so the assignment isn't optimized away) whose `testsuite.sh` synthesizes the design and asserts no `GHDL Bug occurred` crash. (The resulting netlist references a synth-internal `gate_mdff` primitive, which isn't meant to be re-analyzed as standalone VHDL, so this test checks for successful, crash-free synthesis rather than re-analyzing the netlist — unlike the simpler `testsuite/synth/issueNNNN` tests elsewhere in this pass.) This locks in the current, correct behavior as a regression guard.

closed 2026-08-20 by tgingold