ghdl/ghdl #1952: ghdl fails to build: Using -fPIC only for half the build

seijikun, 2022-01-18:
**Description**
Hey guys. I just tried packaging ghdl for openSUSE. My first attempts on this can be seen here: https://build.opensuse.org/package/show/home:seijikun/ghdl
Unfortunately, I can't get ghdl (0e46300c437e39747ebe65e6eae5a131477d28c9) to build. I am seeing very similar errors as have already been reported in this issue: https://github.com/ghdl/ghdl/issues/1400
Unfortunately, the solution from that issue does not seem to work for me.

[Here](https://build.opensuse.org/build/home:seijikun/openSUSE_Tumbleweed/x86_64/ghdl/_log) is a complete build log from the build server.
For build-commands and installed dependencies, you can have a look at the [spec-file](https://build.opensuse.org/package/view_file/home:seijikun/ghdl/ghdl.spec?expand=1) I used.
The build-commands essentially amount to:
```
./configure --prefix=/usr
make -O -j16 V=1 VERBOSE=1
```
(Adding `--default-pic` to the configure call did not change anything for me, unfortunately)

From the build-log, it seems like `-fPIC` is correctly added to gcc invocations in the beginning, but as soon as `gnatmake` is involved, the `-fPIC` option is suddenly missing.

**How to reproduce?**
You can either branch [my attempts](https://build.opensuse.org/package/show/home:seijikun/ghdl) on the OpenBuildService and make changes to the "ghdl.spec" file, or tell me to try different things.

I'm sorry if I did anything horribly wrong!

Paebbels, 2022-01-18:
@seijikun does it mean, GHDL will be generally available in SUSE based operating systems when we get this issue fixed?

umarcor, 2022-01-18:
It seems there are other users building GHDL on opensuse: https://build.opensuse.org/search?search_text=ghdl. Did you check those?

seijikun, 2022-01-18:
> It seems there are other users building GHDL on opensuse: https://build.opensuse.org/search?search_text=ghdl. Did you check those?

openSUSE's OBS instance is open for all distributions. Most of the packages in the search result actually don't build at all, or build for Ubuntu and Debian. There is one working build for openSUSE at the moment, [here](https://build.opensuse.org/package/show/home:javispedro:fpgatools/ghdl).

They fixed the build by injecting the following arguments to make:
```
make GNAT_BARGS="-bargs -E -shared"
```
But since I don't really know much about ada, I dismissed this as a quick workaround.
I thought I had included that in my original post - but I just saw I missed that. Sorry for that!

tgingold, 2022-01-18:
Can you copy the build log ?  It is not available to people outside.

seijikun, 2022-01-18:
> Can you copy the build log ? It is not available to people outside.

Sure, here is an overview:
```
./configure --prefix=/usr
make -O -j16 V=1 VERBOSE=1
```
-> [log](https://github.com/ghdl/ghdl/files/7889420/build.log)

```
./configure --prefix=/usr --default-pic
make -O -j16 V=1 VERBOSE=1
```
-> [log](https://github.com/ghdl/ghdl/files/7889566/build_default_pic.log)

```
./configure --prefix=/usr --default-pic --disable-werror --enable-libghdl --enable-synth
make -O -j16 V=1 VERBOSE=1
```
-> [log](https://github.com/ghdl/ghdl/files/7891907/build_success_with_workaround.log) (this one works - the errors are packaging mistakes)

tgingold, 2022-01-18:
Thanks.

The link errors are due to libgnat.a, which is provided by the compiler.  So it looks this library is not correctly built for OpenSuse.

The workaround (using -bargs -shared) is ok.