ghdl/ghdl #1954: Save all generics in a readable format

Astiolo, 2022-01-19:
Viewing the GHDL sim waveforms is great, but sometimes it's important to see what the generics are too and it's cumbersome to add a print on each one you could be interested in. 

I see that they can't be included in a vcd / ghw file (#438).

A simple alternative could be an option to save them to another file that's human readable. Perhaps something like json, which could be viewed in raw text but could also be opened with many viewers that can make it easy to drill down into the ones of most interest. It would also allow other programs to easily ingest the information if anyone wanted to add that capability.

umarcor, 2022-01-19:
Maybe that's something to be provided through libghdl/pyGHDL.dom?

Astiolo, 2022-01-20:
I don't use pyGHDL, so I don't know how that would fit in.

tgingold, 2022-01-20:
I can try to dump them through ghw, but I am not sure how to display them (gtkwave could be used, but waves aren't the best way to display generics).

Astiolo, 2022-01-20:
I think having them visible in gtkwave would be good, even though they don't change. Then it's all in the one place when you're trying to debug. 

Alternatively, from a user perspective I think it would make sense to be able to include them when using the --disp-tree option. That can already show the ports so including generic values in there would make sense to me.

umarcor, 2022-02-13:
@Astiolo, libghdl is a shared library that provides a subset of GHDL's features; and pyGHDL is the Python interface to that shared library. pyGHDL.dom combines GHDL's parsing/analysis functionality with [vhdl/pyVHDLModel](https://github.com/vhdl/pyVHDLModel) in order to provide an abstract representation of your sources/design:

[image]

As you can see in the [example](https://github.com/vhdl/pyVHDLModel#examples), pyGHDL.dom allows printing the generics (identifiers, modes and subtypes):

```py
from pathlib import Path
from pyGHDL.dom.NonStandard import Design, Document

sourceFile = Path("example.vhdl")

design = Design()
library = design.GetLibrary("lib")
document = Document(sourceFile)
design.AddDocument(document, library)

for entity in document.Entities:
  print("{}".format(entity.Identifier))
  print("  generics:")
  for generic in entity.GenericItems:
    print("  - {} : {!s} {}".format(
      ", ".join([str(i) for i in generic.Identifiers]), generic.Mode, generic.Subtype)
    )
```

Then, you can save that to JSON, YAML or any other format you wish. You can also explore them with some GUI:

[image]

Note that, at the moment, providing the elaborated values of the generics is not supported in libghdl/pyGHDL.dom/pyVHDLModel. You can get everything about them, except the value. That's why I proposed it above.

/cc @Paebbels