ghdl/ghdl #1955: -Wruntime-error not reported for assignments of mismatched length when passed through unsigned()

avelure, 2022-01-25:
**Description**
When one side of an assignment has a different vector length than the other there is normally produced a -Wruntime-error. However, if the assigning signal is passed through a type cast function like std_logic_vector() or unsigned(), there is no error reported.

**Expected behaviour**
Generate the `value constraints don't match target ones [-Wruntime-error]` error.

It would also be nice if the expected and actual values could be added to the error message. But that's more of a feature request.

**How to reproduce?**
```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
  signal data       : unsigned(11 downto 0);
  signal read_data  : std_logic_vector(15 downto 0);
  subtype t_data    is std_logic_vector(11 downto 8);
begin
  read_data(t_data'range) <= std_logic_vector(data);
  data                    <= unsigned(read_data(t_data'range));
end architecture beh;
```

```sh :image: ghdl/ghdl:bullseye-llvm
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 test
```
**Context**
Please, provide the following information:

- OS: win10
- Origin: nightly 0e46300

tgingold, 2022-01-26:
This warning is emitted only when there is a static subtype mismatch.
In this case, the subtype of `unsigned` or `std_logic_vector` is the unconstrained array and therefore there are no static constraints.

Ok, that's technical speak just to tell this is not implemented but it could be.  I will see what I could do!

avelure, 2022-01-26:
Cool, I think I understand.
I noticed the same in a case where there was an arithmetic like "data <= data_small + 1", but I assumed that might be more difficult to determine before elaboration, though these are all caught during analysis with mentor and aldec.

As the bounds error is only thrown during simulation if the branch is hit, it would be useful to detect it during analysis for linting purposes.

tgingold, 2022-01-28:
For linting purposes, you can also do a synthesis.  Those errors will be detected:

```
test.vhdl:12:30:error: mismatching vector length; got 12, expect 4
  read_data(t_data'range) <= std_logic_vector(data);
                             ^
test.vhdl:13:30:error: mismatching vector length; got 4, expect 12
  data                    <= unsigned(read_data(t_data'range));
                             ^
test.vhdl:9:10:warning: signal "read_data" is never assigned and has no default value
  signal read_data  : std_logic_vector(15 downto 0);
```