ghdl/ghdl #1957: VHDL-2008 - Record with unconstrained vector as input

PhillipGeierHEPIA, 2022-02-01:
**Description**
I have a record with unconstrained std_logic_vectors as input to an entity. The record is initialized in a testbench
that passes the record to the entity (dut).

In this example there is no need for the record's content to be unconstrained. However, I use the same mechanism to describe bus interfaces (AXI-Stream) where it is useful to have an unconstrained vector.

**Expected behaviour**
I would expect ghdl to pass the record from the tb to the dut with the constrained sizes.

I tested this with modelsim and it produced no errors / warnings.

**How to reproduce?**
Here is a MWE:

_test_pkg.vhd_
```vhd :file: test_pkg.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package config_pkg is
	  -- Configuration
	type t_config is record
		param_a   : std_logic_vector;
		param_b   : std_logic_vector;
	end record t_config;
end package config_pkg;
```

_dut.vhd_
```vhd :file: dut.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.config_pkg.t_config;

entity dut is
	port(
		i_config : in t_config(
			param_a(31 downto 0),
			param_b(31 downto 0)
		)
	);
end entity;

architecture arch of dut is
begin
end architecture arch;
```

_tb_dut.vhd_
```vhd :file: tb_dut.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.config_pkg.t_config;

entity tb_dut is
end tb_dut;

architecture behaviour of tb_dut is
	component dut is
		port(
			i_config : in t_config(
				param_a(31 downto 0),
				param_b(31 downto 0)
			)
		);
	end component;

	constant s_config : t_config(
		param_a(31 downto 0),
		param_b(31 downto 0)
	) := (
		std_logic_vector(to_unsigned(34,32)),
		std_logic_vector(to_unsigned(55,32))
	);
begin
	dutlbl : dut
		port map(
			i_config => s_config
		);
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-gcc
ghdl -a --std=08 test_pkg.vhd
ghdl -a --std=08 dut.vhd
ghdl -a --std=08 tb_dut.vhd
```

**Context**
- OS: Debian 11.2
- Origin:
Built from sources: `0e46300c437e39747ebe65e6eae5a131477d28c9`

Built with gnat 10.2.1
gcc sources version 10.2.0

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r974.g0e46300c) [Dunoon edition]
Compiled with GNAT Version: 10.2.1 20210110
Target: x86_64-linux-gnu
/home/geier/Documents/dev/tests/ghdl-test/MWE/
Command line:
/home/geier/tools/libexec/gcc/x86_64-pc-linux-gnu/10.2.0/ghdl1 --ghdl--std=08 -P/home/geier/tools/lib/ghdl/ieee/v08/ -P/home/geier/tools/lib/ghdl/ -quiet -o tb_dut.s tb_dut.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap6.adb:989
Call stack traceback locations:
0x7f227ec647c2 0x7a83c1 0x7a2ae3 0x7a3607 0x7a51c4 0x7ef597 0x7c6d12 0x7c321b 0x7f528b 0x7804c8 0x682948 0xc26612 0x6787de 0x687a6d 0x7f55d9 0x676477 0x7f227e723d08 0x67d7d8 0xfffffffffffffffe
******************************************************************

Execution of /home/geier/tools/libexec/gcc/x86_64-pc-linux-gnu/10.2.0/ghdl1 terminated by unhandled exception
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap6.adb:989
Call stack traceback locations:
0x7f227ec647c2 0x7a83c1 0x7a2ae3 0x7a3607 0x7a51c4 0x7ef597 0x7c6d12 0x7c321b 0x7f528b 0x7804c8 0x682948 0xc26612 0x6787de 0x687a6d 0x7f55d9 0x676477 0x7f227e723d08 0x67d7d8 0xfffffffffffffffe
ghdl: compilation error
```

PhillipGeierHEPIA, 2022-02-01:
And I just found out that changing
```constant s_config : t_config```
to 
```signal s_config : t_config```
in tb_dut.vhd solves the issue.

I will leave the issue open for now as I don't think that this is expected behavior.

bjourne, 2022-03-08:
I ran into the same bug. Was about to open a new issue but then I saw this one. Here is an MWE:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package example is
    type fimp_def_t is record
        n_values_per_in_buffer : integer_vector;
        offsets_per_in_buffer : integer_vector;
    end record;

    constant def_a : fimp_def_t(
        n_values_per_in_buffer(0 to 1),
        offsets_per_in_buffer(0 to 1)
    ) := (
        n_values_per_in_buffer => (others => 0),
        offsets_per_in_buffer => (others => 0)
    );
    type fimp_def_arr_t is array(0 to 0) of fimp_def_t;
    constant fimp_defs : fimp_def_arr_t := (0 => def_a);
end package;

package body example is
end example;
```
```
ghdl -a --std=08 test.vhdl      

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: 11.1.0
Target: x86_64-pc-linux-gnu
/home/bjorn/pubwork/Sylva2/glic_tng/schedule_1221/
Command line:
/usr/bin/ghdl1-llvm --std=08 -P/usr/lib/ghdl/ieee/v08/ -P/usr/lib/ghdl/ -c -fpic -o test.o test.vhdl
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:99 access check failed
Call stack traceback locations:
0x55abd3795d27 0x55abd37a9153 0x55abd37a738a 0x55abd37a76bb 0x55abd37e3a00 0x55abd37e43d7 0x55abd37e3e07 0x55abd37e6615 0x55abd37e66ee 0x55abd381b07d 0x55abd381b4db 0x55abd381b5f4 0x55abd3824d2b 0x55abd380f081 0x55abd380e861 0x55abd380e927 0x55abd37f6e6a 0x55abd385acc5 0x55abd385c841 0x55abd385e163 0x55abd36019e0 0x7fc3bdea430e 0x7fc3bdea43bf 0x55abd3600b3c 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
```

tgingold, 2022-03-09:
You are using ghdl version 1.0.0, which is quiet old.  Many vhdl-2008 issues have been fixed, and I am not able to reproduce the crash.

bjourne, 2022-03-09:
It may be specific to the llvm backend. With the mcode backend it doesn't crash. I'll test it with ghdl 2.0.0 too.

dalex78, 2026-08-20:
# Issue #1957 — Crash on a record with unconstrained elements used as a constrained port type

Analysis by an AI agent.

## High-level summary

A record type whose elements are unconstrained (`std_logic_vector`), used as a constrained entity port subtype — `t_config(param_a(31 downto 0), param_b(31 downto 0))` — crashed the gcc and llvm backends with an internal `ASSERT_FAILURE` in `trans-chap6.adb`. mcode is unaffected: it does not use the `src/vhdl/translate` translator this lives in.

## What is actually wrong in GHDL's implementation

`Translate_Selected_Element` (`src/vhdl/translate/trans-chap6.adb`) asserted that the type info of the field taken from the prefix's record layout equals the type info of the element's declared type:

```ada
if Prefix_Tinfo.S.Rec_Fields /= null then
   F := Prefix_Tinfo.S.Rec_Fields (Pos).Fields (Kind);
   El_Tinfo := Prefix_Tinfo.S.Rec_Fields (Pos).Tinfo;
   pragma Assert (El_Tinfo = Res_Tinfo);
```

The assertion contradicts the function's own documentation, three lines above it:

> RES_TINFO is the type info of the result. EL_TINFO is the type info of the field. They can be different when the record subtype is partially constrained or is complex.

and the note at the top of the function:

> Note: EL can be an element_declaration or a record_element_constraint. It can be an element_declaration even if the prefix is of a record subtype with a constraint on EL.

This design is exactly that case. `t_config` declares `param_a : std_logic_vector` unconstrained, and the port is of a subtype that constrains it to 32 bits. `El` is the base element declaration, so `Res_Tinfo` is unbounded while `El_Tinfo`, read from the prefix layout, is the bounded 32-bit one. They differ, and the assertion fires.

## What the fix does

Removes the assertion and records why the two type infos can differ.

Nothing else is needed, because the body is already written for the mismatch: it builds the fat result from `Res_Tinfo` with the bounds read from the prefix's record layout, and accesses the field itself through `El_Tinfo`. The sibling branch, taken when `Rec_Fields` is null, likewise sets `El_Tinfo` from the base element without asserting. The assertion was left behind by the code that learned to handle this case — verified before removing it rather than deleting it because it was in the way.

## Testing

`testsuite/gna/issue1957/` analyzed the design and stopped there, which cannot distinguish a correct record layout from a wrong one. `chk.vhd` adds an entity with the same constrained port that checks what actually arrives, driven from a constant with known values:

```
chk.vhd:22:5:@1ns:(report note): PASS param_a=34 param_b=55
```

Identical values on mcode, which never used this code path and so serves as an independent oracle.

Full `sanity gna vests synth` pass on mcode, llvm and gcc, and with this fix alone applied to master.

closed 2026-08-24 by tgingold