ghdl/ghdl #1958: Tracking issue: VHDL-2019 support

Xiretza, 2022-02-02:
Just like VHDL-2008, VHDL-2019 brings a lot of new features to the language. While some of them are quite out there and probably hard to implement, there are also a few small quality-of-life improvements.

This issue will serve both as a place for discussion about the implementation of VHDL-2019 features in general, as well as to track the implementation of all the Language Changes.

# [List of language changes](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/VHDL2017)
- [ ] [LCS-2016-001](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_001) "Partially Connected Vectors on Port Map"
- [ ] [LCS-2016-002](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_002) "Allow access and protected type parameters on function interfaces"
- [ ] [LCS-2016-004](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_004) "Allow PT methods with AT and FT parms"
- [ ] [LCS-2016-006a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_006a) "File IO / Textio updates"
- [ ] [LCS-2016-006c](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_006c) "File IO / Directory I/O"
- [ ] [LCS-2016-006e](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_006e) "Allow access to system environment variables (baseline)"
- [ ] [LCS-2016-006f](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_006f) "Expose standard conditional analysis identifiers"
- [ ] [LCS-2016-007](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_007) "Sequential Declaration Regions"
- [ ] [LCS-2016-011](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_011) "Date/Time functions"
- [ ] [LCS-2016-012](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_012) "'IMAGE and TO_STRING for composite types"
- [ ] [LCS-2016-014](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_014) "Composites of Protected Types"
- [ ] [LCS-2016-014a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_014a) "Pointers to Composites of Protected Types"
- [ ] [LCS-2016-015](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_015) "Report Calling Path"
- [ ] [LCS-2016-015a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_015a) "ReportCallingPath - Current File and Line by function"
- [ ] [LCS-2016-016](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_016) "Anonymous Types"
- [ ] [LCS-2016-018](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_018) "Attributes for Enumerated Types"
- [ ] [LCS-2016-018a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_018a) "New attributes - 'index"
- [ ] [LCS-2016-018d](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_018d) "New attributes - 'designated_type"
- [ ] [LCS-2016-019](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_019) "Inferring Constraints from Initial Values for Signals and Variables"
- [ ] [LCS-2016-023](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_023) "Relax Library Requirement on Configurations"
- [ ] [LCS-2016-026c](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_026c) "LongIntegers": #950
- [ ] [LCS-2016-030](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_030) "Garbage Collection"
- [ ] [LCS-2016-032](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_032) "PATH_NAME and Shared Variables and Subprograms"
- [ ] [LCS-2016-033](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_033) "Composition with Protected Types"
- [ ] [LCS-2016-034](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_034) "Protected Types with Generic Clause"
- [ ] [LCS-2016-036a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_036a) "Allow for conditional expressions in a declaration (baseline)"
- [ ] [LCS-2016-041](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_041) "Record Introspection / Type Reflection"
- [ ] [LCS-2016-043](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_043) "Attributes for PSL"
- [ ] [LCS-2016-045a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_045a) "Interface - defines new mode view construct for composite interface objects"
- [ ] [LCS-2016-045c](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_045c) "Interface - 'CONVERSE for a mode view"
- [ ] [LCS-2016-047](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_047) "Protected Type: Shared Variables On Entity Interface"
- [ ] [LCS-2016-049](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_049) "Map Generics On Subprogram Call"
- [ ] [LCS-2016-050](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_050) "API for Assert"
- [ ] [LCS-2016-055a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_055a) "Syntax regularization - component declarations"
- [ ] [LCS-2016-059](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_059) "Array Type Generics"
- [ ] [LCS-2016-059a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_059a) "Allow ordering on any scalar array - related to Array Type Generics"
- [ ] [LCS-2016-061](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_061) "Conditional compilation"
- [ ] [LCS-2016-071a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_071a) "Extra optional semicolon at the end of interface_list"
- [ ] [LCS-2016-072b](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_072b) "Function Knows Return Vector Size"
- [ ] [LCS-2016-075](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_075) "Closely related record types"
- [ ] [LCS-2016-082](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_082) "Syntax regularization: empty records"
- [ ] [LCS-2016-086](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_086) "All Interface Lists Can Be Ordered"
- [ ] [LCS-2016-094a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_094a) "Conditional Return Statement"
- [ ] [LCS-2016-099](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_099) "Extended Ranges"

Xiretza, 2022-02-02:
To get the ball rolling, the first thing should probably be to add a `--std=19` option. @tgingold, if it's ok to treat `Vhdl_19` as equivalent to `Vhdl_08` to begin with, I can add that - but I can't check every place where `Vhdl_Std` is used for correctness, especially since I don't actually have a 2019 LRM or an accurate redline.

tgingold, 2022-02-03:
Yes sure.  There are other simple changes that could be done, like adding the new reserved identifier (keywords).

Xiretza, 2022-02-04:
FYI, building the 2019 std library fails with these errors:

```
../../src/std/v19/env.vhdl:213:28:error: variable interface parameter are not allowed for a function (use a constant)
  impure function DIR_OPEN(Dir : out DIRECTORY; Path : in STRING) return DIR_OPEN_STATUS;
                           ^
../../src/std/v19/env.vhdl:213:28:error: mode of a function parameter cannot be inout or out
  impure function DIR_OPEN(Dir : out DIRECTORY; Path : in STRING) return DIR_OPEN_STATUS;
                           ^
../../src/std/v19/env.vhdl:214:29:error: type of constant interface "dir" cannot be record type "directory"
  procedure       DIR_CLOSE(Dir : in DIRECTORY);
                            ^
../../src/std/v19/env.vhdl:214:29:error: (record type "directory" has an access subelement)
  procedure       DIR_CLOSE(Dir : in DIRECTORY);
                            ^
../../src/std/v19/env.vhdl:244:30:error: type of constant interface "call_path" cannot be record type "call_path_element"
  impure function TO_STRING (call_path : CALL_PATH_ELEMENT ) return STRING;
                             ^
../../src/std/v19/env.vhdl:244:30:error: (record type "call_path_element" has an access subelement)
  impure function TO_STRING (call_path : CALL_PATH_ELEMENT ) return STRING;
                             ^
../../src/std/v19/env.vhdl:249:30:error: type of constant interface "call_path" cannot be array type "call_path_vector"
  impure function TO_STRING (call_path : CALL_PATH_VECTOR; Separator : STRING := "" & LF ) return STRING;
                             ^
../../src/std/v19/env.vhdl:249:30:error: (array type "call_path_vector" has an access subelement)
  impure function TO_STRING (call_path : CALL_PATH_VECTOR; Separator : STRING := "" & LF ) return STRING;
                             ^
../../src/std/v19/env.vhdl:255:30:error: type of constant interface "call_path" cannot be access type "call_path_vector_ptr"
  impure function TO_STRING (call_path : CALL_PATH_VECTOR_PTR; Separator : STRING := "" & LF ) return STRING;
                             ^
```

I'm not sure about the first one; the others are solved by LCS-2016-002.

tgingold, 2022-02-05:
I think these are low hanging fruits, at least for the analyzer.

JimLewis, 2022-02-11:
There are 2019 code examples at https://gitlab.com/synthworks/VHDL_2019.    These are the ones I wrote for the presentations I did with Aldec.   

[LCS-2016-002](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_002) will help with some additional data structures I have planned.   
[LCS-2016-061](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_061) will allow library developers to put in work arounds for features being missing - such as conditionally include extra libraries.

Paebbels, 2023-05-27:
@tgingold is [LCS-2016-071a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_071a) solved by 3160b444?

tgingold, 2023-05-27:
Yes.

Paebbels, 2023-05-27:
I checked the checkbox above. Is there anything else that could be checked? E.g. new packages?

tgingold, 2023-05-27:
Not yet.

stfarley, 2024-02-07:
Just wondering if there was any luck getting the standard libraries to compile.

JimLewis, 2024-02-07:
These would also be exciting items to have:
[LCS-2016-015](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_015) "Report Calling Path"
[LCS-2016-015a](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_015a) "ReportCallingPath - Current File and Line by function"

tgingold, 2024-02-08:
Understood!