ghdl/ghdl #1963: Set default maximum --stop-time

breandan, 2022-02-04:
Today, I discovered an orphaned ghdl process had written a several terabytes-large `.ghw` file to disk overnight. For some reason, a simulation does not automatically stop when the process ends, but repeats ad infinitum unless you set the `--stop-time` flag. Can this be chalked up to user error or do you think it would be wise to set a default maximum stop time to prevent this from happening in the future?

umarcor, 2022-02-04:
It's very difficult to set a default maximum stop time, because VHDL is a programming language that allows describing hardware. 

As a programming language, you can write software that never stops. See, e.g. the daemons or web services. A priori, GHDL cannot know if you are using VHDL as an inherently parallel language replacement for C or Ada.

On the other hand, hardware is always running "forever". As soon as you turn it on, all the gates/FFs are working. The control logic decides when to consider the start and the end. In a SoC, the control can be a CPU interacting with the control registers of an IP-core. In simulation, the controller is the testbench. Therefore, you can (need to) explicitly terminate the simulation when you have achieved your purpose. See, for example: https://github.com/VUnit/vunit/blob/master/examples/vhdl/array_axis_vcs/src/test/tb_axis_loop.vhd#L55.

VUnit by default uses `test_runner_cleanup` to terminate the simulation explicitly. When using VHDL 2008, it is using `std.env.stop`. See https://github.com/VUnit/vunit/blob/master/vunit/vhdl/core/src/stop_body_2008p.vhd. Therefore, you might do the same. For instance, add the following process to your testbench:

```
process
begin
  wait for 50 ms;
  std.env.stop(status);
end process;
```

That is functionally similar to using CLI option `--stop-time`. Hence, you could instead decide to use `--stop-time` by default in all of your scripts/makefiles.

Still, `--stop-time` or `std.env.stop` cannot guarantee that the simulation will terminate. There can be infinite loops that prevent advancing of simulation time. That is effectively a halting problem that GHDL cannot resolve.

On a related note, having orphaned GHDL processes might depend on the backend. When using mcode, it should almost never happen. However, when using LLVM or GCC, GHDL calls a binary. When doing Ctrl+C, you might abort GHDL but not the binary. You can avoid it by calling the binary directly instead of using `ghdl -r`. The disadvantage is that you tailoring the scripts/makefiles to a backend.

umarcor, 2022-02-04:
I forgot to mention that VUnit has a watchdog: http://vunit.github.io/run/user_guide.html#the-vunit-watchdog.

breandan, 2022-02-04:
Thank you for the detailed explanation, this is useful to know. I still wonder whether it would be prudent to provide an option to abort simulation after a certain amount of wall clock time has elapsed and return an exit code to the parent process.

tgingold, 2022-02-05:
If you are on linux, there are utilities to kill a process after a timeout.
I am not sure we need to duplicate this feature.

Also, note that simulation ends if all processes are terminated.  That's the natural way to stop simulation.

umarcor, 2022-02-05:
@breandan, as Tristan said, typically you can handle that from the parent process. See, e.g. `timeout --help`, which is available on many GNU/Linux distributions, on MSYS2, etc. Similarly, pytest and similar test management frameworks in other languages do have watchdog/timeout functionality.

closed 2022-02-05 by breandan