ghdl/ghdl #1966: [Question] What does analysis and elaboration do in ghdl?

ju-sh, 2022-02-11:
Hi. I'm new to EDA and was wondering what the analysis and elaborate steps of vhdl does.

Is it something specific to ghdl?

Don't know if it's still valid, but found [here][2] that analysis phase creates an object file.

Also found [here][1] that elaboration step creates an executable. But I couldn't any file except the work-obj.cf file that seems to get generated in the analysis phase and updated in the elaboration phase.

And is there some place other than github where discussions about ghdl happen? Like a mailing list or something (Is ghdl-discuss@gna.org still active?)

[1]: http://ghdl.free.fr/ghdl/Elaboration-command.html
[2]: http://ghdl.free.fr/ghdl/Analysis-command.html

umarcor, 2022-02-11:
> was wondering what the analysis and elaborate steps of vhdl does.

Analysis and elaboration are steps required for "compiling" any computer language. Those are stages of any compiler.
VHDL is a computer language which can be elaborated in various ways (for simulation, for synthesis, etc.). I proposed an updated overview diagram a few days ago: https://github.com/ghdl/ghdl/pull/1964/files. See also https://umarcor.github.io/SIEAV/VHDL.html (work in progress).

I suggest to read any general book/tutorial about what a compiler is and how it works. On the other hand, read the VHDL Language Reference Manual (LRM). You don't need those in order to use GHDL, but they will help you have a better understanding of the subject ("compilation" of HDL).

> Don't know if it's still valid, but found [here](http://ghdl.free.fr/ghdl/Analysis-command.html) that analysis phase creates an object file.

The updated docs are located at ghdl.github.io/ghdl.
See the Quick Start Guide for Simulation: http://ghdl.github.io/ghdl/quick_start/index.html.
You will find basic explanations about `analyze` and `elaborate` in there.
Particularly, object files are created with LLVM or GCC backends, but not with mcode (because it is an in-memory backend).

> Also found [here](http://ghdl.free.fr/ghdl/Elaboration-command.html) that elaboration step creates an executable. But I couldn't any file except the work-obj.cf file that seems to get generated in the analysis phase and updated in the elaboration phase.

Same issue. LLVM and GCC use a compile and link model, and they do produce an executable binary (or a shared lib if you wish). However, the mcode backend is an in-memory JIT compiler, and it does not produce any object or executable; it runs straightaway.

Check the backend you are using with `ghdl version`.

> And is there some place other than github where discussions about ghdl happen? Like a mailing list or something (Is [ghdl-discuss@gna.org](mailto:ghdl-discuss@gna.org) still active?)

Discussions happen either in these issues (it's absolutely ok to open issues for asking/getting help, as you did here) and in the chat: https://gitter.im/ghdl1/Lobby. There are other rooms, such as https://gitter.im/vhdl/General, which might also be interesting for you. I don't think there is any currently active mailing-list.

Xiretza, 2022-02-11:
> Analysis and elaboration are steps required for "compiling" any computer language. Those are stages of any compiler. VHDL is a computer language which can be elaborated in various ways (for [...]

To expand on this, "analysis" roughly corresponds to "compilation", and "elaboration" to "linking". Analysis and Elaboration are terms found in the VHDL Language Reference Manual.

> Discussions happen either in these issues (it's absolutely ok to open issues for asking/getting help, as you did here) and in the chat: https://gitter.im/ghdl1/Lobby. There are other rooms, such as [...]

Note that these rooms can also be accessed via Matrix as [#ghdl1_Lobby:gitter.im](https://matrix.to/#/#ghdl1_Lobby:gitter.im) and [#vhdl_General:gitter.im](https://matrix.to/#/#vhdl_General:gitter.im), if you prefer.

FatsieFS, 2022-02-13:
> To expand on this, "analysis" roughly corresponds to "compilation", and "elaboration" to "linking". Analysis and Elaboration are terms found in the VHDL Language Reference Manual.

Personally I would consider analysis be more equivalent to preprocessing and parsing in classic compiler terminology and elaboration more with compiling and linking. At least from how I see that GHDL does it.

Paebbels, 2022-02-13:
@FatsieFS when you look at GHDL's outputs after analysis, you'll find *.o files if LLVM or GCC backend is used :).

closed 2022-02-21 by tgingold