ghdl/ghdl #1967: GHDL crashes during compilation

tasgomes, 2022-02-14:
**Description**

The following code makes GHDL compilation to crash with exception TYPES.INTERNAL_ERROR.

```vhd
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

entity data_generator_3_tb is
end entity data_generator_3_tb;

architecture rtl of data_generator_3_tb is
begin
    main: process
        variable PixelValue_v : unsigned(11 downto 0);
    begin
        PixelValue_v := (11 downto 10 => to_unsigned(3, 2), others => '0');
        wait;
    end process;
end architecture rtl;
```
This line seems to be the culprit: `PixelValue_v := (11 downto 10 => to_unsigned(3, 2), others => '0');`

**Expected behaviour**
This code is accepted by Modelsim. I also don't see anything wrong according to VHDL language. I would expect this code to compile without errors.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

This was compiled using VUnit framework but shall be reproducible stand-alone.

```sh
E:\msys64_ghdl_1_0_0\mingw64\bin\ghdl.exe -a --workdir=D:\fw\sim\vunit_out\ghdl\ghdl\libraries\lib_data_generator --work=lib_data_generator --std=08 -frelaxed --warn-no-hide D:\fw\par\ipi\data_generator\tb\data_generator_3_tb.vhd
```
**Context**
Please, provide the following information:

- OS: Windows 10 Pro
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: [url](https://repo.msys2.org/distrib/x86_64/msys2-x86_64-20210604.exe)
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
Compiling into lib_data_generator: par\ipi\data_generator\tb\data_generator_3_tb.vhd failed
=== Command used: ===
E:\msys64_ghdl_1_0_0\mingw64\bin\ghdl.exe -a --workdir=D:\fw\sim\vunit_out\ghdl\ghdl\libraries\lib_data_generator --work=lib_data_generator --std=08 -PD:\fw\sim\vunit_out\ghdl\ghdl\libraries\vunit_lib -PD:\fw\sim\vunit_out\ghdl\ghdl\libraries\osvvm -PD:\fw\sim\sim_lib\ghdl\unisim\v08 -PD:\fw\sim\sim_lib\ghdl\secureip\v08  -PD:\fw\sim\vunit_out\ghdl\ghdl\libraries\lib_data_generator -frelaxed --warn-no-hide D:\fw\par\ipi\data_generator\tb\data_generator_3_tb.vhd

=== Command output: ===

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r348.g7d5c6c48) [Dunoon edition]
Compiled with GNAT Version: 10.3.0
Target: x86_64-w64-mingw32
D:\fw\
Command line:
E:\msys64_ghdl_1_0_0\mingw64\bin\ghdl1-llvm.exe --workdir=D:\fw\sim\vunit_out\ghdl\ghdl\libraries\lib_data_generator --work=lib_data_generator --std=08 -frelaxed --warn-no-hide -PD:\fw\sim\vunit_out\ghdl\ghdl\libraries\vunit_lib\ -PD:\fw\sim\vunit_out\ghdl\ghdl\libraries\osvvm\ -PD:\fw\sim\sim_lib\ghdl\unisim\v08\ -PD:\fw\sim\sim_lib\ghdl\secureip\v08\ -PD:\fw\sim\vunit_out\ghdl\ghdl\libraries\lib_data_generator\  -PE:/msys64_ghdl_1_0_0/mingw64/lib/ghdl\ieee\v08\ -PE:/msys64_ghdl_1_0_0/mingw64/lib/ghdl\ -c -o D:\fw\sim\vunit_out\ghdl\ghdl\libraries\lib_data_generator\data_generator_3_tb.o D:\fw\par\ipi\data_generator\tb\data_generator_3_tb.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations:
0x7ff7de82be20 0x7ff7de91ad4b 0x7ff7de91b9da 0x7ff7de91e193 0x7ff7de91e33c 0x7ff7de91dc41 0x7ff7de9552ea 0x7ff7de9553f7 0x7ff7de955e36 0x7ff7de956cbe 0x7ff7de9560df 0x7ff7de958993 0x7ff7de958a4f 0x7ff7de95c4e8 0x7ff7de95c865 0x7ff7de95c5d9 0x7ff7de9a0149 0x7ff7de9b3db1 0x7ff7de9b3ecf 0x7ff7de9b7cc1 0x7ff7de9bd8c1 0x7ff7de97706e 0x7ff7de96952c 0x7ff7de9c998d 0x7ff7de9cb40e 0x7ff7de9ccce5 0x7ff7de792702 0x7ff7de7913bf 0x7ff7de7914f4 0x7ff933417032 0x7ff933f4264f
******************************************************************
E:\msys64_ghdl_1_0_0\mingw64\bin\ghdl.exe: compilation error

Compile failed
make: *** [run_simulation] Error 1
```

**Additional context**
None.

umarcor, 2022-02-14:
@tasgomes, can you please try updating the GHDL package in that MSYS2 installation? The latest version is `1.0.0.r964.g70f3c162b-2`: https://packages.msys2.org/base/mingw-w64-ghdl

tasgomes, 2022-02-14:
oh! Seems like I installed this long time ago :). I updated the version and now everything runs fine. Thanks.