ghdl/ghdl #1969: raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:2819

jrchapple, 2022-02-15:
[object code file deletion errors, if any, reported here in build script]

/media/john/2254C7E454C7B8B5/hdl/z80-cpu/components.vhdl:596:56:error: no declaration for "statein"

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (Debian 1.0.0+dfsg-3) [Dunoon edition]
Compiled with GNAT Version: 10.2.1 20210110
Target: x86_64-linux-gnu
/media/john/2254C7E454C7B8B5/hdl/testbenches/tb-accumulator/
Command line:
/usr/bin/ghdl-gcc -m --workdir=work tb_accumulator
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:2819
Call stack traceback locations:
0x7ff5b07957c2 0x46ad30 0x48425f 0x565729 0x565977 0x564eb0 0x565195 0x565977 0x564eb0 0x565195 0x565977 0x564eb0 0x565c55 0x56dd55 0x57df1e 0x4f45c6 0x409d17 0x7ff5b0254d08 0x409d88 0xfffffffffffffffe
******************************************************************

**Expected behaviour**
does not crash after detecting a coding error

**How to reproduce?**
**Context**
[files.zip](https://github.com/ghdl/ghdl/files/8074118/files.zip)

build is a script that compiles the vhdl code.  It ends with the above exception.
Please, provide the following information:

- OS: Debian bullseye
- ghdl-gcc is from debian bullseye software repository

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
above

**Additional context**
Chose to include all files in build just as they are.  Didn't want to risk removing the cause of the exception or the events that led up to it.
 
I don't know if this bug has been reported previously or has been fixed in a newer version of GHDL.

Cheers
John

tgingold, 2022-02-16:
I cannot reproduce the crash.
Did you try with a recent version of ghdl ?  It is possible that the bug has been fixed meanwhile.

jrchapple, 2022-02-17:
I'm wondering where I can find a more recent version than 1.0.0, which is what I used.
I would prefer not to compile a nightly version, but if that is what it takes (other than fixing my code), so be it.

Paebbels, 2022-02-17:
Compiling on Debian is quite easy for mcode and llvm backends. Do you need the GCC backend variant or are the other also acceptable?

Building mcode backend: https://ghdl.github.io/ghdl/development/building/mcode.html  
Building llvm backend: https://ghdl.github.io/ghdl/development/building/LLVM.html

I'm not sure what GHDL versions are used by Debian in there variants:
* bullseye (stable): 1.0.0+dfsg-3
* bookworm (testing): 1.0.0+dfsg-6
* sid (unstable): 1.0.0+dfsg-8

The -6 variant seams to be created on 31.10.2021: https://packages.debian.org/bookworm/amd64/ghdl-gcc/download

umarcor, 2022-02-17:
@Paebbels, in Debian and other distros they have a somehow strict policy of only distributing tagged versions. That's why during all of 2021, they kept building GHDL 1.0.0. They rebuilt it multiple times, but they didn't change the sources they were cloning. When we tag a new release this month, they will start using that. Therefore, most of the problems we've been having with upstream Debian (and therefore Ubuntu) should be fixed in the following months.

jrchapple, 2022-02-17:
My preference is a gcc back end.

Which version should I be downloading for Debian Bullseye?

umarcor, 2022-02-18:
There is container `ghdl/ghdl:bullseye-gcc-9.1.0`. The corresponding "artifacts" are in `ghdl/pkg:bullseye-gcc-9.1.0`. Hence, you can use the former container, or you can extract the content of the lattter (and install runtime deps, see https://github.com/ghdl/docker/blob/master/run_debian.dockerfile#L34-L40.

closed 2022-03-02 by tgingold