ghdl/ghdl #1970: Runtime simulator dies unexpectedly

MJoergen, 2022-02-16:
**Description**
The runtime simulator dies unexpectedly. How can I debug this further, to determine the root cause and/or find a work-around ?

Could the problem be a lack of resources ? The source files infer memories of size (at least) 8 MegaByte.

The source files consist of a video rescaler that I did not write myself.

Note: The exact same source files simulate successfully in the Vivado simulator.

```
mike@mike-desktop2 ~/git/MJoergen/video_rescaler (main=) $ make
ghdl -i --std=08 --work=work ascal.vhd avm_memory.vhd tb_clk.vhd gen_video.vhd sys.vhd tb.vhd tb.vhd
ghdl -m --std=08 -fexplicit tb
ghdl -r --std=08 tb --assert-level=error --wave=tb.ghw --stop-time=40us
make: *** [Makefile:17: sim] Killed
```

**Expected behaviour**
I expect simulation to proceed and ultimately generate a `ghw` file to view as a waveform.

**How to reproduce?**
See above.

The source files can be provided if needed, but right now I'm looking for help to debug the problem myself, if possible.

**Context**
Please, provide the following information:

- OS: Linux Mint 20.2
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `commit 29289cbb` `Date:   Sat Feb 5 17:05:38 2022 +0100`

```
mike@mike-desktop2 ~/git/MJoergen/video_rescaler (main=) $ ghdl --version
GHDL 2.0.0-dev (1.0.0.r983.g29289cbb) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2022-02-16:
Weird.

Did you check kernel messages (dmesg) ?

You can try to run outside the makefile, you will get more details.
And then you can try to run under gdb to see where it occurs.

Xiretza, 2022-02-16:
My guess would be a stack overflow - see #1592. Try using `--max-stack-alloc=` and `ulimit -s` to confirm this.

MJoergen, 2022-02-16:
* Running outside make gave no new information. It just exits with the string "Killed."
* Running in gdb gave the message `Program terminated with signal SIGKILL, Killed.`. And no, it is not I who killed it.
* Checking `dmesg` gives more information:

```
[299563.477148] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/user.slice/user-1000.slice/user@1000.service,task=ghdl,pid=2569156,uid=1000
[299563.477153] Out of memory: Killed process 2569156 (ghdl) total-vm:64053120kB, anon-rss:62125612kB, file-rss:0kB, shmem-rss:0kB, UID:1000 pgtables:121668kB oom_score_adj:0
[299564.152532] oom_reaper: reaped process 2569156 (ghdl), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
```

So it seems that GHDL is allocating over 64 GB of memory, and is then automatically killed.

I tried monitoring with `top` and could see the amount of virtual memory increase steadily to 60 GB and beyond, so this confirms the above.

Now, I'm surprised GHDL needs more than 64 GB to simulate this design.

The largest memory block is 8 MB, is that too much for GHDL ?

tgingold, 2022-02-16:
Did you try without the --wave option ?

If you file the sources, I can investigate too.

MJoergen, 2022-02-17:
Repeating without the --wave option made no difference. Process still killed after allocating 64 GB.

In an attempt at creating a smaller example to reproduce the error, I removed the video rescaler portion and replaced with a simpler Avalon master module to exercise the 8 MB memory. With this change I still get a failure, but a different one this time. Now I get repeated messages of

```
corrupted double-linked list
corrupted double-linked list
corrupted double-linked list
corrupted double-linked list
make: *** [Makefile:18: sim] Segmentation fault (core dumped)
```

Since this seems easier (?) to debug, I repeated with gdb as follows:

```
mike@mike-desktop2 ~/git/MJoergen/video_rescaler (main=) $ gdb --args ghdl -r --std=08 tb --assert-level=error --stop-time=40us
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ghdl...
(gdb) run
Starting program: /usr/local/bin/ghdl -r --std=08 tb --assert-level=error --stop-time=40us
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff78d7a2f in unlink_chunk (p=p@entry=0x555555ca1890, av=0x7ffff7a2ab80 <main_arena>) at malloc.c:1453
1453	malloc.c: No such file or directory.
(gdb) bt
#0  0x00007ffff78d7a2f in unlink_chunk (p=p@entry=0x555555ca1890, av=0x7ffff7a2ab80 <main_arena>) at malloc.c:1453
#1  0x00007ffff78da773 in _int_malloc (av=av@entry=0x7ffff7a2ab80 <main_arena>, bytes=bytes@entry=152) at malloc.c:4041
#2  0x00007ffff78dc2d4 in __GI___libc_malloc (bytes=152) at malloc.c:3058
#3  0x00007ffff7cc59d1 in __gnat_malloc () from /lib/x86_64-linux-gnu/libgnat-9.so.1
#4  0x00007ffff7cf01ed in system.pool_global.allocate () from /lib/x86_64-linux-gnu/libgnat-9.so.1
#5  0x00005555555d5652 in grt.signals.create_signal (mode=mode_e8, value_ptr=0x7fffd72a6a01, mode_sig=mode_signal, resolv_proc=0x7ffff72b29a4, 
    resolv_inst=(const system__address) 0x0) at /home/mike/git/ghdl/ghdl/src/grt/grt-signals.adb:220
#6  0x00005555555d81c3 in <__ghdl_create_signal_e8> (val_ptr=0x7fffd72a6a01, resolv_func=0x7ffff72b29a4, resolv_inst=0x0)
    at /home/mike/git/ghdl/ghdl/src/grt/grt-signals.adb:1103
#7  0x00007ffff72f8534 in ?? ()
#8  0x00005555559e7eb0 in ?? ()
#9  0x00007fffd72a6010 in ?? ()
#10 0x00007fffd72a6a01 in ?? ()
#11 0x0000008000000071 in ?? ()
#12 0x0000555555ca4900 in ?? ()
#13 0x0040000000000013 in ?? ()
#14 0x0000555555c9fd00 in ?? ()
#15 0x0000555555c9f780 in ?? ()
#16 0x00007f00f72f9a08 in ?? ()
#17 0x0000555555c9f780 in ?? ()
#18 0x00007fffffffd000 in ?? ()
#19 0x00007ffff72fb305 in ?? ()
#20 0x00007f00ffffd000 in ?? ()
#21 0x0000008000000080 in ?? ()
#22 0x0000555555ac9450 in ?? ()
#23 0x0000555555c28b30 in ?? ()
#24 0x0000555555ac9450 in ?? ()
#25 0x0000555555c9f780 in ?? ()
#26 0x00000008559dd8b0 in ?? ()
[... 58 lines cut ...]
in process .tb(simulation).i_avm_master@avm_master(synthesis).P4
  from: work.avm_memory(simulation).DECL_ELAB at avm_memory.vhd:30
  from: work.tb(simulation).i_avm_memory.CMP_ELAB at tb.vhd:21
  from: work.tb(simulation).DECL_ELAB at tb.vhd:36
  from: Elaboration of design
ghdl:error: error during elaboration
==2617173== 
==2617173== HEAP SUMMARY:
==2617173==     in use at exit: 646,697,548 bytes in 1,054,031 blocks
==2617173==   total heap usage: 1,069,915 allocs, 15,884 frees, 664,823,440 bytes allocated
==2617173== 
==2617173== LEAK SUMMARY:
==2617173==    definitely lost: 9,896 bytes in 158 blocks
==2617173==    indirectly lost: 12,164 bytes in 192 blocks
==2617173==      possibly lost: 0 bytes in 0 blocks
==2617173==    still reachable: 646,675,488 bytes in 1,053,681 blocks
==2617173==         suppressed: 0 bytes in 0 blocks
==2617173== Rerun with --leak-check=full to see details of leaked memory
==2617173== 
==2617173== For lists of detected and suppressed errors, rerun with: -s
==2617173== ERROR SUMMARY: 523186 errors from 1 contexts (suppressed: 0 from 0)
mike@mike-desktop2 ~/git/MJoergen/video_rescaler (main=) $ 
```

I will upload the source files for you to try as well.

MJoergen, 2022-02-17:
[tmp.zip](https://github.com/ghdl/ghdl/files/8086029/tmp.zip)

MJoergen, 2022-02-17:
BTW: Looking at my own source code just now it seems the array allocated is of size 64 MB (22 address bits and 128 data bits = 2^22 * 128 bits). This is in fact an error on my part, and the correct value should be 19 address bits, to give a total of 8 MB.

When I change the above minimal example (from 22 address bits to 19 address bits, in lines 11, 93, and 120 of tb.vhd) I get a different behaviour. This time GHDL does not crash, but viewing the process with `top` I see the memory requirements reach approx 23 GB.

So perhaps GHDL (in the original situation) does not gracefully handle out-of-memory errors ?
And is it to be expected that GHDL uses 23 GB to simulate a VHDL signal of 8 MB ? I'm not trying to complain, I'm just interested in knowing the limitations (if any) of the tools I'm using.

umarcor, 2022-02-17:
@MJoergen, note that each bit of a signal needs one byte in memory (because it's a 9-value logic). So, you need to multiply the size by 8, to start with. Furthermore, a signal does not contain the current value only. The language defines attributes and events that require saving the previous values and other metadata. 
I find 23 GB to be too much, but you should expect at least 10x memory requirements for your X MB buffer.

For that reason, it is recommended to use variables instead of signal when modelling memories. That keeps the variable requirements around 8 x X MB. Moreover, some libraries/frameworks use bits instead of std_logic.

tgingold, 2022-02-17:
I can reproduce a crash, I have to investigate.

tgingold, 2022-02-18:
But if `mem` is a variable instead of a signal, I don't get any crash.

tgingold, 2022-02-27:
This is an overflow while computing the memory size to be allocated.
I think this is not the first time it is reported.