ghdl/ghdl #1982: Analysis crashes on large constant array (TYPES.INTERNAL_ERROR)

florg-32, 2022-02-28:
**Description**
When compiling a relatively large constant array (file size: 96KB), the compiler throws an error as shown below.

**How to reproduce?**
See attached file
```sh
ghdl -a --std=08 --work=work bb_mem_pkg.vhd
```

**Context**

- OS: Manjaro stable
- Origin:
  - [x] Package manager: `ghdl-llvm-1.0.0-5` (same results with gcc and mcode)

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: 11.1.0
Target: x86_64-pc-linux-gnu
/media/E/Code/ddca_ss2022_testing/dd/vhdl/decimal_printer/
Command line:
/usr/bin/ghdl1-llvm --std=08 --work=work -P/usr/lib/ghdl/ieee/v08/ -P/usr/lib/ghdl/ -c -fpic -o bb_mem_pkg.o bb_mem_pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations:
0x556a6d40cde8 0x556a6d50860a 0x556a6d5094f9 0x556a6d50c103 0x556a6d50a38a 0x556a6d50a6bb 0x556a6d546a00 0x556a6d546eb8 0x556a6d546df9 0x556a6d549615 0x556a6d5496ee 0x556a6d57e07d 0x556a6d57e4db 0x556a6d57e5f4 0x556a6d587d2b 0x556a6d572081 0x556a6d571861 0x556a6d571927 0x556a6d559e6a 0x556a6d5bdcc5 0x556a6d5bf841 0x556a6d5c1163 0x556a6d3649e0 0x7f536d2c630e 0x7f536d2c63bf 0x556a6d363b3c 0xfffffffffffffffe
******************************************************************
```
[bb_mem.zip](https://github.com/ghdl/ghdl/files/8155950/bb_mem.zip)

tgingold, 2022-02-28:
I cannot reproduce the crash.  You are using an old version of ghdl, you should try with a more recent one.

florg-32, 2022-02-28:
Thanks, after building it myself it compiles

closed 2022-02-28 by florg-32