ghdl/ghdl #1986: synth: exec_name_subtype: cannot handle IIR_KIND_INDEXED_NAME

bewimm, 2022-02-28:
**Description**
The following code (based on #1961) fails to synthesize.

```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity bug is
	port (
		clk : in std_ulogic
	);
end bug;

architecture struct of bug is

	type a_t is record
		value : unsigned;
	end record;

	type a_array_t is array(natural range<>) of a_t;

	type b_t is record
		a : a_array_t;
	end record;

	type b_array_t is array(natural range<>) of b_t;

	function fun(b : b_array_t) return natural is
		variable c : natural;
	begin
		c := 0;
		for i in b'range loop
			for j in b(i).a'range loop
				c := c+1;
			end loop;
		end loop;
		return c;
	end function;

	constant b : b_array_t(0 to 1)(a(0 to 31)(value(31 downto 0))) := (others => (a => (others => (value => (others => '0' )))));
	
	constant dummy : natural := fun(b);
begin
end architecture;
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.10
- Origin:
  - Nightly: `e58709ae`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
exec_name_subtype: cannot handle IIR_KIND_INDEXED_NAME (/home/user/Desktop/ghdl_bug/bug.vhd:30:35)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (v1.0.0-1017-ge58709ae) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/user/Desktop/vhdl_tmp/
Command line:
/home/user/Desktop/ghdl-gha-ubuntu-20.04-gcc/bin/ghdl --synth --std=08 /home/user/Desktop/ghdl_bug/bug.vhd -e bug
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x48f2d2 0x543480 0x54341a 0x545974 0x545a2e 0x5462c9 0x545c70 0x54c0c1 0x550578 0x550590 0x552911 0x552cdf 0x552fc9 0x54257e 0x5451f5 0x54566b 0x544740 0x54500a 0x5835bd 0x4f2f89 0x408b4d 0x7fd23cfa1fce 0x7fd23cfa207b 0x408bac 0xfffffffffffffffe
******************************************************************
```

bewimm, 2022-03-01:
using `for j in b'element.a'range loop` instead of `for j in b(i).a'range loop` leads to a slightly different error:
`sem_selected_name(2): cannot handle IIR_KIND_ELEMENT_ATTRIBUTE (/home/user/Desktop/ghdl_bug/bug.vhd:30:36)`

I can make a separate issue if you prefer that.

tgingold, 2022-03-02:
I think it is worth having a separate issue.  It's invalid.

closed 2022-03-02 by tgingold