ghdl/ghdl #1988: Wrong Version string in GHDL 2.0.0

Paebbels, 2022-03-01:
**Description**
GHDL 2.0.0 was released with a broken version string.

`GHDL 2.0.0 (1.0.0.r1018.g8185ed87) [Dunoon edition]`

**Expected behavior**

```diff
- GHDL 2.0.0 (1.0.0.r1018.g8185ed87) [Dunoon edition]
+ GHDL 2.0.0 [Dunoon edition]
```

umarcor, 2022-03-01:
There are no assets in release 2.0.0: https://github.com/ghdl/ghdl/releases/tag/v2.0.0.

Do you mean you built https://github.com/ghdl/ghdl/commit/8185ed878df5054a9624af97ae4c73bbd3e906ef locally and the version string is the one you showed?

Paebbels, 2022-03-01:
I looked at the CI jobs artifacts for tag 2.0.0 and downloaded e.g. the MinGW64 standalone zip and run `ghdl --version`.

Paebbels, 2022-03-01:
This CI run was running on a tagged version, so only the tag should be visible, but no relative number of commits since last tagged version. This should be displayed only for `***-dev` versions.

Xiretza, 2022-03-01:
I'd say the expected output for the CI run on 8185ed8 would be:

```
GHDL 2.0.0 (2.0.0.r0.g8185ed878) [Dunoon edition]
```

However, the CI run was started before the tag was pushed, so it based its `VER_DESC` off v1.0.0.

Paebbels, 2022-03-01:
That's not correct. CI runs 2 times for the same hash. One as branch-push and one as tag-push.

See:
[image]

So the build in the run of the tag-push can see the tag in the repository history.

-------------
I thought we fixed this problem in one of the previous releases. Hard to tell why it didn't work as expected, because we have not enough releases to test it continuously :(.

umarcor, 2022-03-01:
@Paebbels I'm not sure about CI running twice... I found that sometimes GHA does not repeat the CI of a commit. So, if you push v2.0.0 first, and then push `master` pointing to the same commit, it might ignore running CI for master. Unfortunately, I restarted some of those, so it's not easy to understand what was the sequence.

I built v2.0.0 locally and the version is `2.0.0 (2.0.0.r0.g8185ed878)`, as @Xiretza commented. According to https://github.com/ghdl/ghdl/blob/master/src/version.in, the `()` is always used/shown. Packagers can override the content through `https://github.com/ghdl/ghdl/blob/master/Makefile.in#L185`.

Paebbels, 2022-03-01:
I could accept the `2.0.0 (2.0.0.r0.g8185ed878)` string, but the `GHDL 2.0.0 (1.0.0.r1018.g8185ed87)` is wrong. And that's what we have as an artifact (for hidden downloads).

umarcor, 2022-03-01:
That is wrong probably because the commit was built before pushing the tag, as @Xiretza said. But that's something that only Tristan can know.

Nevertheless, we don't provide artifacts or assets for tagged releases. That's up to packagers. Since last year, we only provide nightly assets.

In this case, the CI of v2.0.0 failed because of an unrelated issue (https://github.com/ghdl/ghdl/commit/3f63e7e87eb5e121526b0e5a07cd5af6ed2855b5), so the release was not created automatically (I did it manually). Anyway, should it had worked, it wouldn't push the assets: https://github.com/ghdl/ghdl/blob/master/.github/workflows/Test.yml#L631-L634. Therefore, it is not critical that the artifacts of that specific CI run have an incorrect version. Those are to be removed when the expiration time passes. We don't expect users to consume them. OTOH, I tested locally that the string is the expected, as long as the tag exists.

tgingold, 2022-03-01:
I tagged locally and then pushed with tags.

umarcor, 2022-03-01:
@tgingold, you pushed the master branch and the tag "at the same time"?

tgingold, 2022-03-01:
Yes.  I used `git push --tags github master`