ghdl/ghdl #1991: synth_dyadic_operation: unhandled IIR_PREDEFINED_ARRAY_LESS_EQUAL

rodrigomelo9, 2022-03-01:
Hi. I am receiving this fault (unhandled IIR_PREDEFINED_ARRAY_LESS_EQUAL) when synthesizing the following (not my real use case, only a test case):

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity issue is
  port (
    clk : in std_logic;
    idx : in std_logic_vector(7 downto 0);
    led : out std_logic
  );
end issue;

architecture implementation of issue is

begin

    process(clk)
    begin
      if (rising_edge (clk)) then
        if idx <= x"10000000" then
          led <= '1';
        else
          led <= '0';
        end if;
      end if;
    end process;

end implementation;
```

```
docker run --rm -v $PWD/..:$PWD/.. -w $PWD hdlc/ghdl:yosys ghdl --synth issue.vhdl -e issue
issue.vhdl:20:16: synth_dyadic_operation: unhandled IIR_PREDEFINED_ARRAY_LESS_EQUAL
        if idx <= x"10000000" then
               ^
issue.vhdl:9:5:warning: no assignment for port "led"
    led : out std_logic
    ^
```

I am using an updated docker image.

rodrigomelo9, 2022-03-01:
It works if I change `if idx <= x"10000000" then` with `if unsigned(idx) <= x"10000000" then` (it also solves the problem in my real situation). So, I guess, the problem is don't abort with an error? (I found this problem trying to synthesize an AXI template from Xilinx, not my code).

tgingold, 2022-03-02:
Yes, this operator can be handled, but it is almost always a mistake.