ghdl/ghdl #1995: altera library doesn't compile under GHDL ... but compiles under questasim/modelsim...

wm2015email, 2022-03-04:
**Description**

altera library doesn't compile under GHDL ... but compiles under modelsim...

**Expected behaviour**

Should compile the same as questasim/modelsim without modifications...

**How to reproduce?**

See, attached files...

PS> wsl ghdl -a --std=02 -frelaxed-rules --ieee=synopsys --warn-no-hide --work=altera_mf altera_mf_components.vhd

PS>  wsl ghdl -a --std=02 -frelaxed-rules --ieee=synopsys --warn-no-hide --work=altera_mf altera_mf.vhd

**ERROR: altera_mf.vhd:31259:24: operator "">="" is overloaded**

[altera_mf.vhd.txt](https://github.com/ghdl/ghdl/files/8188157/altera_mf.vhd.txt)
[altera_mf_components.vhd.txt](https://github.com/ghdl/ghdl/files/8188158/altera_mf_components.vhd.txt)

tgingold, 2022-03-04:
At the end of the error message, you get a tip:

altera_mf.vhdl:31259:24:error: (you may want to use the -fexplicit option)

Paebbels, 2022-03-08:
@wm2015email Please see https://github.com/ghdl/ghdl/tree/master/scripts/vendors. GHDL provides pre-compile scripts to compile all vendor libraries wich matching options.