ghdl/ghdl #1997: Testsuite fails at issue1226 when run in-tree

DanielG, 2022-03-08:
**Description**

When running the test suite right out of the build directory with `make test` gna/issue1226 fails because the -I path is wrong.

**How to reproduce?**

```
$ make
$ make test
[...]
gna issue1226: failed
analyze adder.vhdl
elaborate adder
/usr/bin/gcc
/usr/bin/gcc -c vpi_plugin.c -I/home/user/ghdl/include/ghdl -fPIC
vpi_plugin.c:1:10: fatal error: vpi_user.h: No such file or directory
    1 | #include <vpi_user.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:223: test.mcode] Error 1
```

**Context**

- OS: Debian 11
- Built from sources: git tag `v2.0.0`/311e9448678c6c8ecf1af3dcd7f98c5cd1437c77

DanielG, 2022-03-08:
This was also hit upon in the Fedora packaging issue https://github.com/ghdl/ghdl/issues/888#issuecomment-526881910. In Debian we also have to do a nasty workaround to get the testsuite working because of this (and maybe other issues).

DanielG, 2022-03-08:
Changes:
- Fix offending commit reference, 66cd5e0aa897b947533d269535fde4c0852472c2 is the culprit 786ca17cea32431349608a37b48b608b0e60048f just refactored it.

tgingold, 2022-03-08:
Did you try to do `make install.vpi.local` before running the testsuite ?

DanielG, 2022-03-12:
Ah, that seems to do the trick.

I don't see this being a prerequisite for running the test suite documented
anywhere though? The only reference to that target is in GCC.rst and that's
not even talking about building tests.

Would it make sense to add a dependency on it for the test target(s)?

tgingold, 2022-03-12:
Well, it depends whether you want to run the testsuite after installation or before installation.

Any PR is always welcome!

DanielG, 2022-03-13:
Seems to me unconditionally running the install.vpi.local wouldn't prevent you from running the test suite after installation though? So I'll shoot a PR that adds the dependency.

DanielG, 2022-10-11:
Fixed by #2004, closing.

closed 2022-10-11 by DanielG