ghdl/ghdl #2000: grt-all target broken due to libbacktrace?

DanielG, 2022-03-12:
Summary:`make grt-all` doesn't work and I don't see how it can.

In `grt/Makefile.inc` we have:

```
ifneq ($(GRT_LIBBACKTRACE),)
$(libghdldirsuffix)/$(GRT_LIBBACKTRACE): $(LIBBACKTRACE)
	cp $< $@
endif
```
(`grt-all` then depends on this target directly)

in my tree `$(LIBBACKTRACE)` gets set to `$prefix/$libghdldirsuffix/libbacktrace.a` as defined in the configure script.

This causes the copy to fail since $prefix is some absolute system directory and obviously ghdl isn't installed quite yet during the build :)

tgingold, 2022-03-12:
How did you configure ghdl ?  libbacktrack is optional.

DanielG, 2022-03-12:
Just `./configure`, but I ran grt-all by hand since I'm trying to replace an explicit `make` call to get libgrt.a built in the debian packaging. See https://salsa.debian.org/electronics-team/ghdl/ghdl/-/blob/master/debian/rules#L141 if you want the full context.

tgingold, 2022-10-11:
In a case of a build with gcc, you must install the gcc part before building grt.

DanielG, 2022-10-11:
That simply wont work for distro packaging though. The system parts of the build will only be installed globally when the package is finally installed by the end user.

tgingold, 2022-10-11:
At worst, you can manually copy libbacktrace.a before building grt-all

Any other solution ?