ghdl/ghdl #2008: [ghdl --synth] $fatal not verilog task

suppamax, 2022-03-15:
**Description**
`ghdl --synth --out=verilog` generates code which includes `$fatal` task, which is not a verilog task

**Expected behaviour**
maybe combining `$display` and `$finish`?

**How to reproduce?**
any code which executes
https://github.com/ghdl/ghdl/blob/aff48727a4d0ad4058b7d43a86ebd6ed24b9cc6d/src/synth/netlists-disp_verilog.adb#L1084
or 
https://github.com/ghdl/ghdl/blob/aff48727a4d0ad4058b7d43a86ebd6ed24b9cc6d/src/synth/netlists-disp_verilog.adb#L1097

tgingold, 2022-03-16:
Yes, but it's a systemverilog task.

Use `--no-formal`

suppamax, 2022-03-16:
OK, I just assumed you were targeting plain verilog, not SV.

tgingold, 2022-03-20:
assertions in the netlist is useful for formal verification, hence the use of SV specific routines.
But if you want a pure verilog output, use `--no-formal`.

closed 2022-03-20 by tgingold