ghdl/ghdl #2010: [ghdl --synth] --out to write to file

suppamax, 2022-03-17:
It would be nice if the output of the synthesis could be written to a file instead of to stdout.

This cannot simply be achieved by redirecting the output of `ghdl --synth --out`, because the output also contains other messages which do not belong to the converted module's code.

I know the result could be achieved by using ghdl-yosys-plugin, but this would require using Yosys, but this seems overkill.

tgingold, 2022-03-17:
Which messages do you get in the output ?
`--out` is used in the testsuite using redirection.  Maybe there is one (or more) unexpected message, but it is expected to work.

marzoul, 2024-09-03:
I also would see it very convenient to have an option to specify the ouptut file with a command-line option, as an alternative to the stdout redirection. For the following reasons and from usage experience :
- the output is essentially meant to be stored in a file, for this purpose the stdout redirection is very unnatural
- when using Makefiles the redirection is written in the recipe line, so when doing some debugging by adding command `gdb --args` at beginning of that line, we don't have the output of gdb, thus more edits of command lines needed
- the output file is created by the shell instead of by ghdl, so that file is created even when ghdl stops early on input vhdl errors. So for next commands, `make` thinks that the recipe has succeeded. For complicated Makefiles it is cumbersome to have ghdl-specific recipe coding style to workaround these itches. Or for manual runs with many errors in early stages of development this requires to manually remove the verilog file each time.

marzoul, 2024-09-06:
Additional items to support specifying an output file :
With stdout redirection as vhdl output, logging gets more complicated especially when other tools are involved.
The "normal" messages of ghdl are emitted in stderr which is unusual to log.
Monitoring RAM/time usage by launching ghdl under `/usr/bin/time` places the resource summary in same log than ghdl, thus more cumbersome to manually examine or filter, needs more regex etc.

closed 2024-09-08 by tgingold

marzoul, 2024-09-08:
:star_struck:

commit 3790d926 (2024-09-08): synth: add option -o=FILE to write to a file

Fix #2010