ghdl/ghdl #2011: synth: setting only top 32 bits of a 64 bit variable fails

antonblanchard, 2022-03-18:
I'm chasing a synthesis issue in the Microwatt FPU. A cut down test case:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity testcase is
    port (
        misc_sel : in std_ulogic_vector(3 downto 0);
        result   : out std_ulogic_vector(63 downto 0)
        );
end entity testcase;

architecture behaviour of testcase is
begin
    testcase_0: process(all)
        variable misc : std_ulogic_vector(63 downto 0);
    begin
        case misc_sel is
            when "1101" =>
                misc := x"FFFFFFFF00000000";
                --misc := x"FFFFFFFF80000000";
            when others =>
                misc := x"0000000000000000";
        end case;
        result <= misc;
    end process;
end architecture behaviour;
```

Built with:

```
ghdl --synth --std=08 testcase.vhdl -e testcase
```

Shows both legs of the case as all zeroes:

```
  with n4_o select n7_o <=
    "0000000000000000000000000000000000000000000000000000000000000000" when '1',
    "0000000000000000000000000000000000000000000000000000000000000000" when others;
```

If I set one of the low 32 bits of misc (`FFFFFFFF80000000`), then we see the correct behaviour:

```  with n4_o select n7_o <=
    "1111111111111111111111111111111110000000000000000000000000000000" when '1',
    "0000000000000000000000000000000000000000000000000000000000000000" when others;
```

tgingold, 2022-03-18:
Oops sorry, you're right.

closed 2022-03-18 by tgingold

antonblanchard, 2022-03-18:
Thanks @tgingold! I verified this fixes the original issue.

commit f1b1d83f (2022-03-18): synth-vhdl_context: adjust mask.  Fix #2011