ghdl/ghdl #2013: synth: Issue when conditionally setting bits within a vector

antonblanchard, 2022-03-19:
I have the following test case:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity testcase is
    port (
        state   : in std_ulogic;
        class   : in std_ulogic;
        o       : out std_ulogic_vector(3 downto 0)
        );     
end entity testcase;

architecture behaviour of testcase is
    signal misc_sel      : std_ulogic_vector(3 downto 0);
begin
    testcase_0: process(all)
    begin 
        misc_sel <= "0000";

        case state is
           when '0' =>
                misc_sel <= "0111";
           when '1' =>
                if class = '1' then
                    misc_sel(3) <= '1';
                end if;
           when others =>
        end case;
    
        o <= misc_sel;
    end process;
end architecture behaviour;
```

And the following test bench:

```
entity tb_testcase is
end tb_testcase;

library ieee;
use ieee.std_logic_1164.all;
    
architecture behav of tb_testcase is
  signal state : std_ulogic;
  signal class : std_ulogic;
  signal o     : std_ulogic_vector(3 downto 0);
begin
  dut: entity work.testcase
    port map (state, class, o);
    
  process
  begin
    state <= '0';
    class <= '0';
    wait for 1 ns;
    assert o = "0111";

    state <= '1';
    class <= '0';
    wait for 1 ns; 
    assert o = "0000";

    state <= '1';
    class <= '1';
    wait for 1 ns;
    assert o = "1000";
    
    wait;
  end process;
end behav;
```

It works if I run the test bench:

```
ghdl -c --std=08 tb_testcase.vhdl testcase.vhdl -e tb_testcase
./tb_testcase
```

But if I pass the testcase through synthesis it fails:

```
ghdl --synth --std=08 testcase.vhdl -e testcase > testcase-synth.vhdl
ghdl -c --std=08 tb_testcase.vhdl testcase-synth.vhdl -e tb_testcase
./tb_testcase
tb_testcase.vhdl:20:5:@1ns:(assertion error): Assertion violation
```

If I look at the synthesized verilog, it looks like we get (at least) 2 things wrong. We have two selects, one for the bottom
3 bits of misc_sel and one for the top bit. The constant for bottom 3 bits is using the top 3 bits, and the constant for the top bit
is using the bottom bit:

```
  with n9_o select n12_o <=
    "000" when "10",
    "011" when "01",               -- should be "111"?
    "000" when others;

  with n9_o select n15_o <=
    n6_o when "10",
    '1' when "01",                    -- should be '0'?
    '0' when others;
  n19_o <= n15_o & n12_o;             -- 1 & 011 (was expecting 0111)
```

closed 2022-03-20 by tgingold

antonblanchard, 2022-03-20:
Thanks @tgingold, I confirmed this does fix my issue.

antonblanchard, 2022-03-21:
@umarcor would it be possible to kick off a build of the GHDL synthesis docker container (I think its hdlc/ghdl:yosys) with this (and the other fix a few days ago)?

umarcor, 2022-03-21:
@antonblanchard give me a few hours to trigger everything...

umarcor, 2022-03-21:
@antonblanchard, it done now!

```sh
$ docker run --rm -it hdlc/ghdl:yosys ghdl version
GHDL 3.0.0-dev (v2.0.0-51-gd77a17b1) [Dunoon edition]
...
``

antonblanchard, 2022-03-21:
@umarcor Thank you! I verified that fixes all my synthesis issues (working on another sky130 tape out)

tgingold, 2022-10-11:
Thank you for the status update.  Weird it has not been already caught.

commit 62337bdc (2022-03-20): synth-vhdl_expr(value2logvec): fix vlen handling.  Fix #2013