ghdl/ghdl #2015: Precompiling vendor primitives (Xilinx Vivado) to use with GHDL

p-kuzmanovic, 2022-03-25:
**(Approach #1)**: ghdl-0.33, vendor scripts from `'master'`
I have GHDL 0.33 (pre-build binary) on Windows, so followed [this guide](https://ghdl.github.io/ghdl/getting.html) to install Xilinx Vivado primitives. I get an error:

[image]

I assume that `'-Wbinding'` option for command `-a` is a more recent feature, so I figured I should update GHDL.

**(Approach #2)**: ghdl-2.0.0
I followed [this guide](https://ghdl.github.io/ghdl/development/building/mcode.html#build-mcode) to build GHDL from sources (mcode backend). I installed dependencies like GNAT GPL and zlib. After this, I get this error when trying to run `make`:

[image]

I tried with both GNAT 64 and 32 and the error message is the same/similar.

Thank you, any help is much appreciated!

Paebbels, 2022-03-25:
At first you're mixing 2 different problem reports in one issue.

**For the first:** GHDL v0.33 is very old. Currently, GHDL 2.0.0 was release some days ago and the latest development version is v3.0.0-dev. Please install GHDL v2.0.0 or v3.0.0-dev and run the vendor scripts from such a version.

GHDL has changed and added options / renamed options as well as the external libraries have added files or changed compile orders.

**For the second:** You're building GHDL from within MinGW64, but calling the GNAT compiler. Please use the ADA compiler provided by MinGW64 and if targeting the LLVM backend, also the LLVM provided by MinGW64. In such a case, please follow the instructions as written in our documentation. That's mainly calling `configure`, `make` and finally `make install`.

If no option is provided, mcode backend is assumed.

p-kuzmanovic, 2022-03-28:
Thanks Patrick. I managed to compile GHDL 2.0.0 and the Xilinx primitives using [MSYS2](https://www.msys2.org/). I attach the console output, I assume it is successful even though it reports some "SCRIPT ERRORS". If you can confirm this, I guess we can close this issue.

[image]

However, I now have an issue now when trying to run GHDL executable (error code 0xc000007b). I can open another issue for this.

Paebbels, 2022-03-28:
I think the filter scripts needs some updates due to modified output formats from GHDL.

That's why the scripts report them, so someone could adjust the filtering.

You can disable filtering with a command line switch. See the online documentation or `<script> --help` for details.