ghdl/ghdl #2016: synth error: actual signal must be a static name

DanielG, 2022-03-28:
**Description**
I'm trying to build a register interface that allows for automatically assigned addresses and reflecting/converting the raw bit values in the register file into a record type containing varying field types. I thought I finally found a way to do it without repeating myself a lot or generating VHDL code with a script but I get stuck at an error that I think shouldn't be happening with VHDL2008.  

Everything works as expected up to the point where I try to introduce a procedure call to handle_field in a loop in the `convert_regs_records` procedure which triggers an error:

```
ent.vhdl:68:35:error: actual signal must be a static name
      handle_field(records.config3(i));
                                  ^
```

Note that the loop itself doesn't seem to be a problem it's using the loop variable `i` instead of a constant that breaks it.

**Expected behaviour**
The code emitting this warning references `LRM93 2.1.1.2`, the VHDL2008 LRM doesn't have this section anymore the equivalent seems to be `4.2.2.3 Signal parameters`. My reading of it doesn't seem to mandate any sort of check for staticness but I'm not very familliar with the nomenclature here.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package pkg is
  subtype regword_t is unsigned(7 downto 0);
  type reg_array_t is array (0 to 10) of regword_t;

  type records_t is record
    config1 : bit;
    config2 : bit;
    config3 : bit_vector(3 downto 0);
  -- ...
  end record;
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.pkg.all;

entity ent is
  port (
    records : out records_t;
    regs    : out reg_array_t;
    wr_data : in regword_t;
    wr_en   : in std_ulogic;
    addr    : in natural
    );
end entity ent;

architecture a of ent is
  -- Overloaded type conversions
  function from_regword (regword : regword_t) return bit is
  begin
    if regword(0) then return '1';
    else return '0'; end if;
  end function;
  function to_regword (value : bit) return regword_t is
  begin
    if value = '1' then return (0 => '1', others => '0');
    else return (others => '0'); end if;
  end function;

  procedure convert_regs_records(
    signal records : inout records_t;
    signal regs : out reg_array_t;
    signal wr_data : in regword_t;
    signal wr_en : std_ulogic;
    signal addr  : natural)
  is
    variable i : natural := 0;
    variable iregs : reg_array_t := (others => (others => '0'));

    procedure handle_field (signal f : inout bit) is
    begin
      iregs(i) := to_regword(f);
      if wr_en='1' and addr = i then
        f <= from_regword(wr_data);
      end if;
      i := i + 1;
    end procedure;
  begin
    handle_field(records.config1);
    handle_field(records.config2);

    for i in records.config3'range loop
      handle_field(records.config3(i));
      --^ error: actual signal must be a static name
      -- Note: Replacing i with 0 works though.
    end loop;

    regs <= iregs;
  end procedure;

begin
  convert_regs_records(records, regs, wr_data, wr_en, addr);
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl synth --std=08 ent.vhdl -e
```

**Context**
- OS: Debian bullseye
- Origin: Custom Debian package built from 2.0.0 git tag.

diogratia, 2022-03-29:
*The code emitting this warning references LRM93 2.1.1.2, the VHDL2008 LRM doesn't have this section anymore the equivalent seems to be 4.2.2.3 Signal parameters. My reading of it doesn't seem to mandate any sort of check for staticness but I'm not very familliar with the nomenclature here.*
.
It's not a warning, it's an error:

4.2.2.3 Signal parameters

>If an actual signal is associated with a signal parameter of any mode, the actual shall be denoted by a static signal name. ...

8.Names  
8.1 General  

>A name is said to be a static name if and only if one of the following conditions holds:  
>>— The name is a simple name or selected name (including those that are expanded names) that does not denote a function call, an object or value of an access type, or an object of a protected type and (in the case of a selected name) whose prefix is a static name.  
>>  
>>— The name is an indexed name whose prefix is a static name, and every expression that appears as part of the name is a static expression.  
>  
>...    
>A *static signal name* is a static name that denotes a signal. ,,,

1.3 Structure and terminology of this standard  
1.3.1 General

>In this document, the word *shall* is used to indicate a mandatory requirement. The word *should* is used to indicate a recommendation. The word *may* is used to indicate a permissible action. The word can is used for statements of possibility and capability.

1.3.3 Semantic description  

>The meaning and restrictions of a particular construct are described with a set of narrative rules immediately following the syntactic productions. In these rules, an italicized term indicates the definition of that term, and identifiers appearing entirely in uppercase letters refer to definitions in package STANDARD (see 16.3).  
>  
The following terms are used in these semantic descriptions with the following meanings:  
>**erroneous:** The condition described represents an ill-formed description; however, implementations are not required to detect and report this condition. Conditions are deemed erroneous only when it is impossible in general to detect the condition during the processing of the language.  
>  
>**error:** The condition described represents an ill-formed description; implementations are required to detect the condition and report an error to the user of the tool.  
>  
>**illegal:** A synonym for “error.”  
>  
>**legal:** The condition described represents a well-formed description.

A description not meeting a semantic requirement (the quoted sentence from 4.2.2.3, the use of the mandatory requirement shall)  is not well formed. It's either an error or erroneous, because it's detectable an error.

Well formed has a dictionary meaning "constructed according to grammatical rules" (grammatical - well formed; in accordance with the productive rules of the grammar of a language) where the grammar ( a set of actual or presumed prescriptive notions about correct use of a language) for VHDL includes lexical, syntactical and semantic rules.

tgingold, 2022-10-11:
But in 4.2.2.3, there is (at the 6th paragraph):

If an actual signal is associated with a signal parameter of any mode, the actual shall be denoted by a static signal name.

jeras, 2024-01-29:
We encountered a similar issue in our testbench code, we solved it by replacing signals in the procedures with variables. Variables do not trigger the same error. Where the procedures were used we added temporary variables used to convert signals into variables and used the variables as arguments to the procedures. In your case you would probably need another conversion from temporary variable back into a signal for procedure outputs. Since you do not have any clock or timing code in the procedure, I think this approach would work.

We also checked in other simulators on EDA playground, whether this is specific to GHDL, and it is not, the same error was reported by other simulators when using signals, but using variables was fine. The procedures with signals worked well with the Vivado simulator, this is why we were not sure whether this was a GHDL issue or a VHDL rule.