ghdl/ghdl #2017: Porting GHDL makefile from v0.33 to v2.0.0

p-kuzmanovic, 2022-03-28:
I managed to build GHDL 2.0.0 from sources with MSYS2 (mcode code generator, compiled with GNAT 11.2.0). My makefile header looks like this:
```
GHDL=ghdl
GHDLFLAGS= --std=08 --work=work --workdir=work

all :
	if exist "work" rmdir /Q /S work
	mkdir work
```

It worked with v0.33, but now I get the following error:
```
if exist "work" rmdir /Q /S work
/bin/sh: -c: line 2: syntax error: unexpected end of file
make: *** [Makefile:5: all] Error 2
```
[image]

Xiretza, 2022-03-28:
This has nothing to do with ghdl itself, but rather your makefile, which contains code that isn't a valid shell script. It looks like it might be windows batch commands, but I don't know enough about that language to tell. I think what you want is the following:

```make
all :
	rm -rf work
	mkdir work
```

p-kuzmanovic, 2022-03-28:
Thanks, this worked! It now goes through, but I get a segmentation fault and I can't view my waveform with GTKWave. I guess this is now related to the issue #2018

[image]

umarcor, 2022-03-28:
mcode backend is not supported on Windows 64 bit. For simulation purposes, you need to use the LLVM backend or build for 32 bit (MINGW32, instead of MINGW64).

p-kuzmanovic, 2022-03-28:
Got it! I managed to build with LLVM and MINGW64 now. Now I get the following error:

```
C:\msys64\usr\local\bin\ghdl.exe: installation problem: cc not found
make: *** [all] Error 1
```

[image]

umarcor, 2022-03-28:
When using it for simulation, GHDL is a compiler. It analyses VHDL and generates machine code, which needs to be linked. Therefore, it requires a C compiler at runtime. Since you used MINGW64 to build GHDL, you might want to use it within MSYS2's MINGW64 shell, instead of using the CMD. Otherwise, you will need to make some compiler available in the environment. That might be the one from MSYS2, or some other.

Paebbels, 2022-03-28:
@p-kuzmanovic
You can either:
* use the GHDL built with MinGW64 inside of MinGW64, so it will find MinGW's GCC binaries and libs, or * you need to set en environment variable, so MinGW64 paths are visible outside of MinGW64 e.g. in CMD or PowerShell, or
* you copy the needed executables, DLLs, *.a, *.o files into the GHDL install ordner relative to ghdl.exe.

I could help with the latter setup, because I use GHDL in such a standalone variant, but variant 1 is the simplest for beginner. If you want to try option 3, please contact me e.g. in Gitter.