ghdl/ghdl #2020: Xilinx Aurora 8B/10B elaboration erorr

p-kuzmanovic, 2022-03-29:
I've been trying to simulate my design that includes Xilinx Aurora 8B/10B core. I find the alternatives (ModelSim Started, Xilinx XSIM) either very slow or cumbersome to use in script mode, so I'm trying GHDL to see if I can get improvement. I managed to analyse all required files from Aurora output products.

[image]

Next step is to elaborate the entities, but I get the following error:

```
ghdl -e --std=93c --work=work --workdir=work -fsynopsys -PC:/PLDesign/GHDL-sim/xilinx-vivado/unisim/v93 aurora_8b10b_0_reset_logic
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/PLDesign/GHDL-sim/xilinx-vivado/unisim/v93\unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__numeric_std__ELABORATED' does not match section name '//AArdLC'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/PLDesign/GHDL-sim/xilinx-vivado/unisim/v93\unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__std_logic_1164__ELABORATED' does not match section name '//AArdLv'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/PLDesign/GHDL-sim/xilinx-vivado/unisim/v93\unisim_retarget_VCOMP.o:ortho:(//AArdLC[.refptr.ieee__numeric_std__ELABORATED]+0x0): multiple definition of `.refptr.ieee__numeric_std__ELABORATED'; C:\msys64\usr\local\lib/ghdl\ieee\v93\numeric_std-body.o:ortho:(.rdata$.refptr.ieee__numeric_std__ELABORATED[.refptr.ieee__numeric_std__ELABORATED]+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/PLDesign/GHDL-sim/xilinx-vivado/unisim/v93\unisim_retarget_VCOMP.o:ortho:(//AArdLv[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): multiple definition of `.refptr.ieee__std_logic_1164__ELABORATED'; C:\msys64\usr\local\lib/ghdl\ieee\v93\std_logic_1164-body.o:ortho:(.rdata$.refptr.ieee__std_logic_1164__ELABORATED[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
C:\msys64\usr\local\bin\ghdl.exe: compilation error
make: *** [Makefile:61: all] Error 1
```

tgingold, 2022-03-30:
That's not expected.  Those are messages from the linker.
Can you try on Linux ?
If not, are you able to create a testcase you could submit ?

p-kuzmanovic, 2022-03-30:
I stripped down the Makefile and zipped the testcase, together with Aurora sources and library dependencies. I hope you can reproduce the error. I use MINGW64 from MSYS2.

[issue#2020-testcase.zip](https://github.com/ghdl/ghdl/files/8378034/issue.2020-testcase.zip)

p-kuzmanovic, 2022-03-30:
I tried on Linux/Ubuntu and elaboration of top level goes through, with a lot of `-Wbinding` warnings. I believe that primitives/components like `IBUFDS_GTE2`, `BUFG` and others are a part of UNISIM. Am I doing something wrong with my `-e` command?

[image]

Paebbels, 2022-03-30:
depends on what signals aurora needs from common and channel.
if it's 1-channel aurora, you could write a component taking 32-bit user input, serialize it to the P/N lines and on the other side you unpack to 32-bits. This would be a stub model for just transmitting the data. Maybe you need some kind of rx/tx_resetdone signals for the core too.

Paebbels, 2022-03-30:
I tied once the post-synthesis model, but it was to big for GHDL with all logic being broken down to LUTs and FFs. it was several MB of a VHDL file.

tgingold, 2022-04-01:
So the bug doesn't reproduce on Linux.  I will investigate on Windows when possible.

amb5l, 2022-08-26:
I am seeing the same error, I think. I'm using MSYS2 on Windows, GHDL version is:
```
GHDL 3.0.0-dev (2.0.0.r394.g4afeced94) [Dunoon edition]
 Compiled with GNAT Version: 12.1.0
 llvm code generator
```

I'm not using Aurora - just unisim:

```
ghdl -e --std=08 -frelaxed np6532_functest
D:/devtools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__numeric_std__ELABORATED' does not match section name '//AArdLD'
D:/devtools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__std_logic_1164__ELABORATED' does not match section name '//AArdLw'
D:/devtools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: unisim_retarget_VCOMP.o:ortho:(//AArdLD[.refptr.ieee__numeric_std__ELABORATED]+0x0): multiple definition of `.refptr.ieee__numeric_std__ELABORATED'; D:\devtools\msys64\mingw64\lib/ghdl\ieee\v08\numeric_std-body.o:ortho:(.rdata$.refptr.ieee__numeric_std__ELABORATED[.refptr.ieee__numeric_std__ELABORATED]+0x0): first defined here
D:/devtools/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: unisim_retarget_VCOMP.o:ortho:(//AArdLw[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): multiple definition of `.refptr.ieee__std_logic_1164__ELABORATED'; D:\devtools\msys64\mingw64\lib/ghdl\ieee\v08\std_logic_1164-body.o:ortho:(.rdata$.refptr.ieee__std_logic_1164__ELABORATED[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
```
To reproduce, run `make elaborate` in the `build/np6532_functest_ghdl` directory of [this repo](https://github.com/amb5l/tyto2) (note: submodules required). External dependancies include Xilinx Vivado 2022.1 and the ca65 assembler and ld65 linker from the [cc65 toolchain](https://cc65.github.io/).