ghdl/ghdl #2026: Problem with generic package

std-max, 2022-04-07:
**Description**
I have an internal error when running the following code. It uses a generic package, but I am not sure whether it is the reason of the crash.

**Expected behaviour**
Simulation should run without any errors. Works fine with Modelsim.

**How to reproduce?**
Here is a small MWE:

```vhd :file: test_tb.vhd
package test_pkg is
    type bv_rec_t is record
        bv : bit_vector;
    end record;
end package;

-------------------------------------------------------------------------------

library work;
use work.test_pkg.all;

package test_constrained_generic_pkg is
    generic (
        BV_WIDTH : positive
    );
    subtype bv_subrec_t is bv_rec_t(bv(BV_WIDTH-1 downto 0));
end package;

-------------------------------------------------------------------------------

package test_constrained_pkg is new work.test_constrained_generic_pkg generic map (BV_WIDTH => 4);

-------------------------------------------------------------------------------

library work;
use work.test_pkg.all;

entity dut is
    port (
        input : in bv_rec_t
    );
end entity;

architecture rtl of dut is
begin
end architecture;

-------------------------------------------------------------------------------

library work;
use work.test_constrained_pkg.all;

entity test_tb is
end entity;

architecture tb of test_tb is
    signal bv0 : bv_subrec_t;
begin
    i_dut : entity work.dut
        port map (
            input => bv0
        );
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 test_tb.vhd
ghdl -e --std=08 test_tb
ghdl -r --std=08 test_tb
```

**Context**
- OS: `Debian GNU/Linux 10 (buster)`
- GHDL: `GHDL 2.0.0-dev (1.0.0.r909.g48dc5595.dirty)`

**Bug box**

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r909.g48dc5595.dirty) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
Command line:
ghdl -e --std=08 test_tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Call stack traceback locations:
0x55ddd626316e 0x55ddd626354d 0x55ddd62637c6 0x55ddd647c318 0x55ddd647bef8 0x55ddd648b4e5 0x55ddd648b940 0x55ddd6435463 0x55ddd64359eb 0x55ddd643c35d 0x55ddd643c3bb 0x55ddd6441496 0x55ddd6430821 0x55ddd649ab45 0x55ddd63ff16d 0x55ddd63fc27f 0x55ddd640a8a2 0x55ddd649f1bb 0x55ddd63dd071 0x55ddd6310a46 0x55ddd64a2b4f 0x55ddd60a1b55 0x7f17f2887099 0x55ddd60a01d8 0xfffffffffffffffe
******************************************************************
```

umarcor, 2022-04-08:
I can confirm (using `gcr.io/hdl-containers/ghdl`):

```
$ ghdl -e --std=08 test_tb 

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (v2.0.0-77-g4775ebbd) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/src/
Command line:
ghdl -e --std=08 test_tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Call stack traceback locations:
0x555f2ab5d46f 0x555f2ab5d84f 0x555f2ab5dacb 0x555f2ad769a2 0x555f2ad76582 0x555f2ad85c1f 0x555f2ad8607a 0x555f2ad2fdd6 0x555f2ad30354 0x555f2ad36c7b 0x555f2ad36cd9 0x555f2ad3bceb 0x555f2ad2b1e3 0x555f2ad954bb 0x555f2acfa305 0x555f2acf7458 0x555f2ad05680 0x555f2ad99c61 0x555f2acd816c 0x555f2ac0bc19 0x555f2ad9d5d1 0x555f2a99cbb5 0x7f9f53a9cd08 0x555f2a99b208 0xfffffffffffffffe
******************************************************************
```

tgingold, 2022-04-15:
Very interesting catch.
I am testing a fix for it.
Thank you very much for the small reproducer.

closed 2022-04-15 by tgingold

std-max, 2022-04-15:
Thank you @tgingold  !

commit 61c76d59 (2022-04-15): trans-chap2.adb: adjust owner_subtype for instantiations.  Fix #2026