ghdl/ghdl #2028: GHDL Bug occurred: cannot handle IIR_KIND_EXTERNAL_SIGNAL_NAME .

YaroslavZazulyak2022, 2022-04-08:
**Description**
I am trying to compile my test bench with this command in GHDL:
ghdl -m --std=08 --work=work  -Pxilinx-vivado -P/  -fsynopsys -frelaxed hash_tb

and  this feedback from GHDL :
analyze ../../../Example_project/to_yz_private/to_vhdl_X_prj/hash_vhdl/hash_vhdl_sim.srcs/sources_1/imports/hash_v08/hash_tb.vhd
translate_expression: cannot handle IIR_KIND_EXTERNAL_SIGNAL_NAME (../../../Example_project/to_yz_private/to_vhdl_X_prj/hash_vhdl/hash_vhdl_sim.srcs/sources_1/imports/hash_v08/hash_tb.vhd:91:13)

In my test bench on line 91 I am used this VHDL v2008 constraction to allocate path to signal somewhere in my submodules:
 bus_1_w <= << signal .hash_tb.uut.u_map.u_map_module_1.module_2.bus_1_in1 : std_logic_vector(state_size_c - 1 DOWNTO 0) >>;

This construction VHDL v2008 with the existing path is working well in Vivado but did not in GHDL.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 11.2.0
Target: x86_64-w64-mingw32
C:\GHDL\work_ghdl\work_hash_v08\
Command line:
C:\GHDL\bin\ghdl1-llvm.exe --std=08 --work=work -fsynopsys -frelaxed -Pxilinx-vivado\ -PC:/msys64/\ -PC:\GHDL\lib/ghdl\ieee\v08\ -PC:\GHDL\lib/ghdl\ -c -o hash_tb.o ../../../Example_project/to_yz_private/to_vhdl_X_prj/hash_vhdl/hash_vhdl_sim.srcs/sources_1/imports/hash_v08/hash_tb.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x7ff6b50dbbac 0x7ff6b520a7e8 0x7ff6b5209943 0x7ff6b525fdc1 0x7ff6b525ff2c 0x7ff6b52606f0 0x7ff6b52611d2 0x7ff6b52635fc 0x7ff6b5263735 0x7ff6b5267569 0x7ff6b526d870 0x7ff6b5224323 0x7ff6b52167fb 0x7ff6b5279db2 0x7ff6b527b995 0x7ff6b527d2dc 0x7ff6b50326e8 0x7ff6b50313af 0x7ff6b50314e4 0x7ff8ca6d7032 0x7ff8cb1c264f
...
******************************************************************
```

tgingold, 2022-04-10:
Yes, external names are not yet supported.

tgingold, 2023-09-17:
Now supported (with mcode)

closed 2023-09-17 by tgingold