ghdl/ghdl #2029: GHDL crash

TopiRinkinen, 2022-04-09:
Hi the following simple code crashes GHDL

```
ghdl --version
GHDL 3.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tester is
        generic(
                data_bits: natural
        );
        port(
                clk_in: in std_logic;
                a_reset_in: in std_logic;
                wr_ready_out: out std_logic;
                wr_valid_in: in std_logic;
                wr_data_in: in std_logic_vector(data_bits-1 downto 0);
                rd_ready_in: in std_logic;
                rd_valid_out: out std_logic;
                rd_data_out: out std_logic_vector(data_bits-1 downto 0);
                err_out: out std_logic
        );
end;

architecture formal of tester is
begin
        default clock is rising_edge(clk_in);
        assume always
        -- assert always
                {not a_reset_in; wr_valid_in and not wr_ready_out} |=>
                {stable(wr_valid_in) and stable(wr_data_in)};
end;
```

```
ghdl -a --std=08 tester.vhdl
translate_expression: cannot handle IIR_KIND_PSL_STABLE (tester.vhdl:28:42)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
/mnt/c/work/formal/clock_adaptor/formal/rtl/vhdl/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -fpic -c -o tester.o tester.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x4ba1ce 0x606de3 0x5cc88a 0x5cca26 0x5ccaef 0x5cdf4b 0x5d0a77 0x6518ed 0x59fc89 0x658992 0x65a3a3 0x65b9ed 0x40b46b 0x7fa78a5540b1 0x40a5dc 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
```

Environment: WSL in Win10

TopiRinkinen, 2022-04-09:
Noticed that the crash happens in llvm backend, but not in mcode backend.

tmeissner, 2022-04-09:
PSL functions like stable() are only implemented for synthesis at the moment. See #1616