ghdl/ghdl #2030: Support for multiple libraries in ghdl-ls

Farmadupe, 2022-04-09:
The [documentation](https://github.com/ghdl/ghdl-language-server#usage) for ghdl-lsp does not describe a mechanism for mapping design files into libraries. From looking at the [python source code](https://github.com/ghdl/ghdl/blob/33043febf136c4d1096b83885ec5823b8c2bb843/pyGHDL/lsp/workspace.py#L208) for the language server I think there is no such undocumented feature.

Without this capability, the language server cannot be compatible with any codebase with multiple libraries (3rd-party or vendor IP).

tgingold, 2022-04-09:
By default, libraries are read by ghdl-ls.  So you can simply analyze the files before starting the editor.

But yes, it would be nice to support any library in the project file.

Farmadupe, 2022-04-09:
Thanks. To enable this workaround, I assume I should dump GHDL's post-analysis output files into the workdir defined in the project json file?

tgingold, 2022-04-10:
No, that's not required.  You could also use option -P.