ghdl/ghdl #2036: ghdl-ls crashes when hdl-prj.json has a source source file that contains more than one design unit

Farmadupe, 2022-04-13:
**Description**
When using ghdl-ls to analyze, the server executable crashes when analyzing any source file with more than one design unit.

**How to reproduce?**
This minimal case elicits the error for me:

```vhd :file: pkg_1.vhd
package a is end;
package b is end;
```

```json :file: hdl-prj.json
{"files": [{ "file": "pkg_1.vhd", "language": "vhdl" }]}
```

Representative stack traces (line numbers may differ because I annotated ghdl-ls python with some print statements):
```
Traceback (most recent call last):
  File "C:\Users\t\AppData\Local\Programs\Python\Python310\lib\site-packages\pyGHDL\lsp\lsp.py", line 122, in handle
    response = fmethod(**params)
  File "C:\Users\t\AppData\Local\Programs\Python\Python310\lib\site-packages\pyGHDL\lsp\vhdl_ls.py", line 112, in textDocument_documentSymbol
    doc = self.workspace.get_or_create_document(textDocument["uri"])
  File "C:\Users\t\AppData\Local\Programs\Python\Python310\lib\site-packages\pyGHDL\lsp\workspace.py", line 107, in get_or_create_document
    res = self.create_document_from_uri(doc_uri)
  File "C:\Users\t\AppData\Local\Programs\Python\Python310\lib\site-packages\pyGHDL\lsp\workspace.py", line 100, in create_document_from_uri
    sfe = document.Document.load(source, os.path.dirname(path), os.path.basename(path))
  File "C:\Users\t\AppData\Local\Programs\Python\Python310\lib\site-packages\pyGHDL\lsp\document.py", line 51, in load
    sfe = files_map.Reserve_Source_File(dirid, fileid, buf_len)
  File "C:\Users\t\AppData\Local\Programs\Python\Python310\lib\site-packages\pyGHDL\libghdl\_decorator.py", line 192, in inner
    raise LibGHDLException(
pyGHDL.libghdl.LibGHDLException: Caught exception when calling 'files_map__reserve_source_file' in libghdl.

Traceback (most recent call last):
  File "C:\Users\t\AppData\Local\Programs\Python\Python310\lib\site-packages\pyGHDL\libghdl\_decorator.py", line 189, in inner
    return functionPointer(*args)
OSError: [WinError 541541187] Windows Error 0x20474343

```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
OS: Windows 11
ghdl version: v3.0.0-dev. Some recent LLVM prebuild for windows (downloaded within the last week)
ghdl-language-server: [Built from master revision f28b61](https://github.com/ghdl/ghdl-language-server/commit/f28b612b5585f713a6d966fff726dfebae0d6078)

Farmadupe, 2022-04-13:
After a little more experimentation, the error may actually be due to the _existence_ of hdl-prj.json. The language server does not crash while analyzing ```pkg_1.vhd``` when hdl-prj.json does not exist

Farmadupe, 2022-04-15:
After some more investigation, it seems that this issues is not related to the number of design units in a single file.

I think that the root cause is that when hdl-prj.json exists on windows, (lib)ghdl can be presented with URIs for the same file but with different byte strings, which causes some internal error inside the ada portion of ghdl itself. Here's an example of different representations being presented to GHDL. The first line is sourced from hdl-prj.json. The second line is sourced from the VS code.
```
2022-04-15 12:05:27,986 [DEBUG] qqqqqqqqqcreating document: URI: file:///c:/Users/t/Desktop/ghdl-test/proj/pkg_1.vhd
2022-04-15 12:05:28,000 [DEBUG] qqqqqqqqqcreating document: URI: file:\c%3A\Users\t\Desktop\ghdl-test\proj\pkg_1.vhd
```

I assume that possible fixes would be 1: to change the ada portion of libghdl to accept any resolvable URI, or 2: Find a way to normalize windows paths inside the python language server such that the Ada code only ever sees one single representation of file URIs. I assume the latter option is easiest. 

The differences in the example are: Quoting, and slash direction. Also if the ada portion of the GHDL is case sensitive for filenames (I didn't read far enough in the code to check) then this should be considered as well. The resulting URIs should probably be compliant with [RFC8089](https://datatracker.ietf.org/doc/html/rfc8089#appendix-E) 

So perhaps the following solution might work:
For each URI that is to be sent from python to Ada:
1) Unquote the uri
2) replace backslashes in the path portion with forward slashes

tgingold, 2022-04-15:
Ah, so this is a windows specific issue.
How did you start ghdl-ls ?
Can you set GHDL_LS_TRACE=1 in the environment before it starts ?  This will create log files.

Farmadupe, 2022-04-15:
'm fairly sure that Ada function files_map::reserve_source_file is (wrongly?) called twice for the same vhdl source file . 

The first time (no crash) is at server startup when reading the hdl-prj.json file, the python stack trace is something like workspace::read_project() -> workspace::read_files_from_project ->workspace::add_vhdl_file -> Document::load -> files_map::reserve_source_file. The exact text passed to ada is derived only from the hdl-prj.json file (and the current working directory)

The second time (with crash), files_map::reserve_source_file is called when the source file is opened in the text editor, the stack trace is vhdl_ls.py::textDocument_didOpen -> workspace.py::put_document -> put_document -> create_document_from_uri -> Document::load -> files_map::reserve_source_file

There is a check in workspace.py::put_document that I think the intention is to prevent this error, but I assume it is failing due to differences in string representation of file names in windows:
```
doc = self.get_document(doc_uri)
        if doc is None:
            doc = self.create_document_from_uri(doc_uri, source=source, version=version)
```

Hence I assume that a valid fix is to somehow 'canonicalize' windows paths before entering the ada portion of GHDL

tgingold, 2022-04-15:
I think you're right.
lsp.path_from_uri is supposed to canonicalize windows paths.

(I don't use windows, so there might be cracks)

Farmadupe, 2022-04-15:
From reading the python I think there are many bodge/hack fixes that can be applied. 

But I think there is a risk that such a bodge fix might leave some edge case crashes (for example.. crashing if the user used an uppercase "C:" in a windows drive name if the user puts an absolute path in hdl-prj.json)

closed 2022-08-09 by Farmadupe