ghdl/ghdl #2038: GHDL crashes after using an external name to alias a shared variable of a hierarchy instance with an access type

aritz-alonso, 2022-04-17:
**Description**

I have a testbench that instantiates an entity capable of generating time based waveforms. The entity has a shared variable that contains an access type to control the shape of the waveform. I'm trying to alias this shared variable in the testbench using an external name, but GHDL crashes every time I try to run the testbench.

**Expected behaviour**
GHDL crashes as soon as the simulation is started.

**How to reproduce?**
I've grouped the vhdl sources for the modulator in a library named 'modulation_lib'. This library contains four source files:

* types_pkg.vhd: contains a type with the different symbols the waveform generator can produce.
```vhd :file: types_pkg.vhd
package types_pkg is
    type t_symbol is (NARROW_SYMBOL, REGULAR_SYMBOL, WIDE_SYMBOL);
end package;
```

* generic_stack_pkg.vhd: a protected type stack that will hold the symbols to be created by the waveform generator. It is a generic stack adapted from the code found in https://vhdlwhiz.com/linked-list/
```vhd :file: generic_stack_pkg.vhd
-- Adapted from https://vhdlwhiz.com/linked-list/
package generic_stack_pkg is
  generic (
    type datatype
  );
  type stack is protected
    procedure push(data : in datatype);
    impure function pop return datatype;
    impure function is_empty return boolean;
  end protected;
end package generic_stack_pkg;

package body generic_stack_pkg is
  type stack is protected body
    type item;
    type ptr is access item;
    type item is record
      data : datatype;
      next_item : ptr;
    end record;
    variable root : ptr;

    procedure push(data : in datatype) is
      variable new_item : ptr;
      variable node : ptr;
    begin
      new_item := new item;
      new_item.data := data;
      if root = null then
        root := new_item;
      else
        node := root;
        while node.next_item /= null loop
          node := node.next_item;
        end loop;
        node.next_item := new_item;
      end if;
    end;

    impure function pop return datatype is
      variable node : ptr;
      variable ret_val : datatype;
    begin
      node := root;
      root := root.next_item;
      ret_val := node.data;
      deallocate(node);
      return ret_val;
    end;

    impure function is_empty return boolean is
    begin
      return root = null;
    end;
  end protected body;
end package body generic_stack_pkg;
```

* symbol_stack_pkg,vhd: a concrete instance of generic_stack_pkg.vhd using the symbol type t_symbol found in types_pkg.vhd
```vhd :file: symbol_stack_pkg.vhd
library modulation_lib;
use modulation_lib.types_pkg.all;
package symbol_stack_pkg is new modulation_lib.generic_stack_pkg generic map (datatype => t_symbol);
```

* modulator.vhd: the waveform generator itself
```vhd :file: modulator.vhd
library ieee;
use ieee.std_logic_1164.all;

library modulation_lib;
use modulation_lib.types_pkg.all;

entity modulator is
  port (
    req_modulation : in std_logic;
    busy : out std_logic;
    pwm : out std_logic
  );
end modulator;

architecture behavioral of modulator is
  shared variable symbol_buffer : modulation_lib.symbol_stack_pkg.stack;
begin
  p_modulate : process
    variable v_pwm_symbol : t_symbol;
  begin
    busy <= '0';
    pwm <= '0';
    main_loop : loop
      wait until req_modulation = '1';
      busy <= '1';
      while not symbol_buffer.is_empty loop
        v_pwm_symbol := symbol_buffer.pop;
        pwm <= '1';
        case v_pwm_symbol is
          when NARROW_SYMBOL =>
            wait for 10 us;
          when REGULAR_SYMBOL =>
            wait for 30 us;
          when WIDE_SYMBOL =>
            wait for 50 us;
          when others =>
            assert false report "Unknown modulation symbol" severity error;
        end case;
        pwm <= '0';
        wait for 5 us;
      end loop;
      busy <= '0';
    end loop; -- main_loop
  end process p_modulate;
end architecture behavioral;
```

Finally, the waveform generator is instantiated in the following testbench file:
```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

library modulation_lib;
use modulation_lib.types_pkg.all;

entity testbench is

end entity testbench;

architecture tb of testbench is

    signal tb_modulator_pwm : std_logic;
    signal tb_modulator_req_modulation : std_logic;
    signal tb_modulator_busy : std_logic;

begin

    inst_modulator : entity modulation_lib.modulator(behavioral)
    port map (
        req_modulation => tb_modulator_req_modulation,
        busy => tb_modulator_busy,
        pwm => tb_modulator_pwm
    );

    proc_stimuli : process
        alias v_modulation_buffer is <<variable .testbench.inst_modulator.symbol_buffer : modulation_lib.symbol_stack_pkg.stack>>;
    begin
        report "Simulation start";
        v_modulation_buffer.push(NARROW_SYMBOL);
        v_modulation_buffer.push(NARROW_SYMBOL);
        v_modulation_buffer.push(WIDE_SYMBOL);
        v_modulation_buffer.push(REGULAR_SYMBOL);
        v_modulation_buffer.push(WIDE_SYMBOL);
        tb_modulator_req_modulation <= '1';
        wait until tb_modulator_busy = '1';
        tb_modulator_req_modulation <= '0';
        wait until tb_modulator_busy = '0';
        report "Simulation complete";
        std.env.finish;

    end process proc_stimuli;

end tb;
```

To launch the testbench I use the following GHDL commands:
```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -i --std=08 --work=modulation_lib types_pkg.vhd
ghdl -i --std=08 --work=modulation_lib generic_stack_pkg.vhd
ghdl -i --std=08 --work=modulation_lib symbol_stack_pkg.vhd
ghdl -i --std=08 --work=modulation_lib modulator.vhd
ghdl -i --std=08 --work=work testbench.vhd
ghdl -m --std=08 testbench
ghdl -r --std=08 testbench --assert-level=error --wave=testbench.ghw
```

**Context**

I've tried to execute the testbench using a Windows machine, in Ubuntu 18.04 LTS under WSL and in a Centos8 machine. In all cases, GHDL crashed with a bug report.

Windows bug report:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (v0.37) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: i686-w64-mingw32
D:\Repos\ghdl_crash\sim\run\
Command line:
D:\eda\GHDL\0.37-mingw32-mcode\bin\ghdl.exe -r --std=08 testbench --assert-level=error --wave=../out/testbench.ghw
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap4.adb:1621
Call stack traceback locations:
0x64d740 0x6b37bc 0x6290b6 0x62695f 0x631cef 0x6b7cd5 0x60f314 0x5b0373 0x6ba883 0x402879 0x401361 0x7581fa27 0x77997a7c 0x77997a4c
******************************************************************
```

Ubuntu 18:04 LTS under WSL bug report:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r93.gff5c35e4) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/mnt/d/Repos/ghdl_crash/sim/run/
Command line:
ghdl -r --std=08 testbench --assert-level=error --wave=../out/testbench.ghw
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Call stack traceback locations:
0x55c45b1dfc95 0x55c45b1e00ad 0x55c45b1dbd6d 0x55c45b1dbdd3 0x55c45b3962db 0x55c45b399cec 0x55c45b39f253 0x55c45b39f2bd 0x55c45b3a2741 0x55c45b3a79b5 0x55c45b418e8c 0x55c45b37c64b 0x55c45b3797c5 0x55c45b387f06 0x55c45b41d9f9 0x55c45b358a46 0x55c45b28dd3d 0x55c45b421346 0x55c45b015421 0x7f912a2acc85 0x55c45b013bd8 0xfffffffffffffffe
******************************************************************
```

Centos8 bug report:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r91.gd71a4391) [Dunoon edition]
Compiled with GNAT Version: 8.5.0 20210514 (Red Hat 8.5.0-10
Target: x86_64-redhat-linux
/home/centosuser/ghdl_crash/sim/run/
Command line:
ghdl -r --std=08 testbench --assert-level=error --wave=../out/testbench.ghw
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Call stack traceback locations:
0x5bf970 0x5bfd37 0x5bc0e1 0x5bc147 0x76ad34 0x76e5a0 0x773874 0x7738db 0x776bd9 0x77bb40 0x7e92d8 0x751bc8 0x74ee1d 0x75cf18 0x7edb66 0x72e072 0x668eb7 0x7f108b 0x40ac8f 0x7f54070cbca1 0x40930c 0xfffffffffffffffe
******************************************************************
```

**Additional context**
I have succesfully executed the tesbench above with "Modelsim - Intel FPGA Starter Edition 10.5b (Quartus Prime 17.1)" simulator in a indows machine

diogratia, 2022-04-18:
Analyzing testbench.vhd generates the bug report. Low and behold there's an alias declaration for a variable external name in process proc_stimuli where in external names are not implement as yet in ghdl. See Issues #1471 and #24 where we see external name supports has likely (to quote Robert Downey Jr. as Sherlock Holmes) at the 'contemplative' stage since at least 2016. (External names are not easy to implement, nor have they been of a particularly high priority.)

The different exception location in the ghdl-0.37 can be discounted, that release from February 2020.

aritz-alonso, 2022-04-19:
> Analyzing testbench.vhd generates the bug report. Low and behold there's an alias declaration for a variable external name in process proc_stimuli where in external names are not implement as yet in [...]

Thank you for the reply! I'll modify my testbenches so that they do not rely on external names. I hope this feature is implemented by GHDL at some time.

commit 9619183f (2023-09-18): testsuite/gna: add a test for #2038

Close #2038
closed 2023-09-18 by tgingold