ghdl/ghdl #2039: Transmit a package through a hierarchy of entities

dalex78, 2022-04-19:
**Description**
The following code aims at passing a formal generic package from a generic list to a generic map of an entity. Simply put: transmit a package through a hierarchy of entities.

**Expected behaviour**
No compilation errors.

_This code might be invalid (more details at the end of the issue)_

**How to reproduce?**
MWE:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package axil_generic_pkg is

    generic(
        ADDR_WIDTH : natural
    );

    type mosi_t is record
        valid : std_ulogic;
        addr  : unresolved_unsigned(ADDR_WIDTH-1 downto 0);
    end record;

    type miso_t is record
        ready : std_ulogic;
    end record;

end package;

----------------------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;

entity router_write_channel is
    generic(
        package axil is new work.axil_generic_pkg generic map(<>)
    );
    port(
        axil_si : in  axil.mosi_t;
        axil_so : out axil.miso_t
    );
end entity;

architecture testbench of router_write_channel is
begin
    -- do something
end architecture;

----------------------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;

entity router is
    generic(
        package axil is new work.axil_generic_pkg generic map(<>)
    );
    port(
        axil_si : in  axil.mosi_t;
        axil_so : out axil.miso_t
    );
end entity;

architecture testbench of router is
begin

    router_write_channel: entity work.router_write_channel
    generic map(
        axil => axil
    )
    port map(
        axil_si => axil_si,
        axil_so => axil_so
    );

end architecture;

----------------------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library std;
library work;

entity tb is
end entity;

architecture testbench of tb is

    package axil40 is new work.axil_generic_pkg
    generic map(
        ADDR_WIDTH => 40
    );

    signal axil_si : axil40.mosi_t;
    signal axil_so : axil40.miso_t;

begin

    router: entity work.router
    generic map(
        axil => axil40
    )
    port map(
        axil_si => axil_si,
        axil_so => axil_so
    );

    scenario_process : process
    begin
        wait for 50 ns;
        -- some test
        std.env.finish;
    end process;

end architecture;
```

How to launch and results:
```python
$ ghdl -a --std=08 test.vhd
test.vhd:68:17:error: actual of association is not a package instantiation
        axil => axil
                ^
test.vhd:71:20:error: can't associate 'axil_si' with port "axil_si"
        axil_si => axil_si,
                   ^
test.vhd:71:20:error: (type of 'axil_si' is mosi_t)
test.vhd:35:9:error: (type of port "axil_si" is mosi_t)
ghdl:error: compilation error
```

**Context**
GHDL version:
Test done with the following version using the container `ghdl/ghdl:bullseye-gcc-9.1.0`
```python
GHDL 3.0.0-dev (v2.0.0-93-gff5c35e4) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 GCC back-end code generator
```
OS:
```python
$ cat /etc/os-release | grep ^PRETTY | cut -d"\"" -f2
Debian GNU/Linux 11 (bullseye)
$ cat /etc/debian_version
11.3
```

**Additional context**
This code does not compile either with Modelsim (Inter FPGA Starter Edition 2020.1):

```python
$ vcom -2008 test.vhd
Error: ../test/test_axi_pkg_5/temp.vhd(67): (vcom-1590) Bad expression in actual part of association element 'axil'. 
Error: ../test/test_axi_pkg_5/temp.vhd(72): Actual parameter for interface package declaration 'axil' is not a package instantiation. 
Error: ../test/test_axi_pkg_5/temp.vhd(70): Signal [mode IN port] "axil_si" is type work.router.axil.mosi_t; expecting type work.router_write_channel.axil.mosi_t. 
Error: ../test/test_axi_pkg_5/temp.vhd(71): Signal [mode OUT port] "axil_so" is type work.router.axil.miso_t; expecting type work.router_write_channel.axil.miso_t. 
Note: ../test/test_axi_pkg_5/temp.vhd(74): VHDL Compiler exiting 
End time: 13:10:10 on Apr 17,2022, Elapsed time: 0:00:00 
Errors: 4, Warnings: 0 
```

According to @[Tricky](https://stackoverflow.com/users/8567231/tricky), it does not compile with Questa 2020.2 either.

There is a [StackOverflow](https://stackoverflow.com/questions/71901430/transmit-a-package-through-an-hierarchy-of-entity) post on this exact same code and a few exchange on [Gitter-VHDL](https://gitter.im/vhdl/General?at=625bf819c43500250043c1cd).

To resume:
@[Tricky](https://stackoverflow.com/users/8567231/tricky) and @[JimLewis](https://github.com/JimLewis) think that it might be a valid code (they only done a quick overlook and might have __miss something subtle__).

User [user16145658](https://stackoverflow.com/users/16145658/user16145658) thinks that **the above code is illegal**: [StackOverflow-comments](https://stackoverflow.com/questions/71901430/transmit-a-package-through-an-hierarchy-of-entity#comment127059555_71901430)

tgingold, 2022-10-11:
Indeed, not sure if this is valid.

According to LRM08 6.5.7.2 Generic map aspects:

An actual associated with a formal generic package in a generic map aspect shall be a name that denotes an instance of the uninstantiated package named in the formal generic package declaration [...]

But here, axil denotes a formal generic package.  This formal generic package denotes an instantiated package (according to 6.5.6.2), so it denotes indirectly.

dalex78, 2022-10-11:
Hum, I see. It's a shame :-(
Should I raise an issue on the VASG for this ? Is it worse it ?