ghdl/ghdl #2041: VHDL case-others is not translated to verilog case-default

KasperHesse, 2022-04-20:
**Description**
A VHDL `case` statement with an `others` clause is not translated to a Verilog `case` statement with a corresponding `default` clause when synthesizing to Verilog.

**Expected behaviour**
The generated Verilog `case` statement should have an `others`-clause. Since it does not have this, the generated Verilog and original VHDL are not equivalent when simulating.

**How to reproduce?**
The following VHDL

```vhd :file: mymodule.vhd
library ieee;
use ieee.std_logic_1164.all;

entity mymodule is
  port (
    clk: in std_logic;
    reset: in std_logic;
    i: in std_logic_vector(1 downto 0);
    o: out std_logic
  );
end mymodule;

architecture rtl of mymodule is
signal o_next: std_logic;
begin
process(i) begin
  o_next <= '0';
  case i is
    when "00" => o_next <= '0';
    when others => o_next <= '1';
  end case;
end process;

process(clk, reset) begin
  if rising_edge(clk) then
    if reset = '1' then
      o <= '0';
    else
      o <= o_next;
    end if;
  end if;
end process;
end rtl ;
```
And running
```ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 mymodule.vhd
ghdl -synth --std=08 --out=verilog mymodule
```
generates
```verilog
module mymodule
  (input  clk,
   input  reset,
   input  [1:0] i,
   output o);
  wire o_next;
  wire n4_o;
  reg n7_o;
  wire n14_o;
  reg n17_q;
  assign o = n17_q;
  /* mymodule.vhd:15:8  */
  assign o_next = n7_o; // (signal)
  /* mymodule.vhd:22:5  */
  assign n4_o = i == 2'b00;
  /* mymodule.vhd:21:3  */
  always @*
    case (n4_o)
      1'b1: n7_o <= 1'b0;
    endcase
  /* mymodule.vhd:29:5  */
  assign n14_o = reset ? 1'b0 : o_next;
  /* mymodule.vhd:28:3  */
  always @(posedge clk)
    n17_q <= n14_o;
endmodule
```
where I would expect the case statement to be
```verilog
always @*
  case (n4_o)
    1'b1: n7_o <= 1'b0;
    default: n7_0 <= 1'b1;
  endcase
```

**Context**
Please, provide the following information:

- OS: Ubuntu 18.0.4 running under WSL 2 on Windows 10 v. 19044.1645
- Origin:
  - [x] Built from sources: ` ec6709df668b989f6543a7da985e894c6b538efc`

tgingold, 2022-04-20:
Yes, the output is weird.  I am investigating.

tgingold, 2022-04-21:
Should be fixed.  Looks like it was not fully handled in the code that generates verilog

commit 3c6df8a2 (2022-04-21): netlists-disp_verilog: output default value for pmux.  Fix #2041
closed 2022-04-21 by tgingold