ghdl/ghdl #2044: Add -Werror command line flag

f00fdiv, 2022-04-26:
I like to compile my projects such that a build fails if there are any warnings, thereby forcing me to fix any warnings immediately.
I think it would be nice to have a -Werror switch similar to what compilers such as gcc/clang currently have, which forces any warnings to report as errors and exit with a non-zero status. Maybe I'm blind, but I haven't found any reference to this in the GHDL documentation.

I often automatically build projects with a Makefile. So sometimes the warnings can get lost in verbose outputs, I end up not noticing them until much later when I change something else which causes the output to appear in a different order.

Example of current usage:
`ghdl --synth [module name]` results in `warning: [warning text]` (Exit 0)

Example with proposed -Werror flag
 `ghdl -Werror --synth [module name]` results in `error: [warning text], (-Werror)` (Exit 1)

Or something to that effect.

I'd be happy to take this on myself if it's something the maintainers want to go ahead with.

umarcor, 2022-04-26:
It seems that the feature is supported already: https://github.com/ghdl/ghdl/search?q=Werror
Maybe you need to write the option *after* the command: `ghdl --synth -Werror ...` or `ghdl synth -Werror ...`.

umarcor, 2022-04-26:
Nevertheless, it is not shown in the docs. Therefore, even though it might exist, it is worth keeping this issue open to update the docs.

f00fdiv, 2022-04-26:
It looks like you are correct, adding `-Werror` immediately _after_ `--synth` results in the expected behaviour.
Would we want to also support its use before the command i.e. `ghdl -Werror --synth ...`?

umarcor, 2022-05-12:
I dont think so. Arguments provided to GHDL need to be located in one of two locations, depending on the type:

`elab-run <[options...] [library.]top_unit [arch] [simulation_options...]>`
http://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-elab-run

`synth <[options...] file... -e [top_unit [arch]]>`
http://ghdl.github.io/ghdl/using/Synthesis.html#cmdoption-ghdl-0

The `[options]` that GHDL can accept depend on the subcommand. IMHO, supporting some of those options before the subcommand would be cumbersome to maintain and it would provide no much benefit (users do still need to understand that options and simulation options are different).

See #1204.

closed 2022-06-17 by tgingold