ghdl/ghdl #2047: "ghdl --version" reports "tarball" when built from git repo source

LukasVik, 2022-05-02:
**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `01c330faa`

# Background 

With recent versions of git (see more below), the output from `ghdl --version` reports "tarball" even when GHDL is built from a git repo checkout. I used to utilize this version string for version control of some compiled artifacts, which is now impossible since it always reports the same ("tarball").

```
master [/home/lukas/work/repo/ghdl/ghdl]$ ghdl --version
GHDL 3.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

After running ``./configure`` and ``make``, the contents of ``version.tmp`` reads as:

```
--  This file is automatically generated by Makefile

package Version is
   Ghdl_Ver     : constant String :=
    "3.0.0-dev";

   Ghdl_Ref     : constant String :=
     "master";

   Ghdl_Hash    : constant String :=
     "01c330faaed81ecd4a38bfcdc4838ea0f518e5d7";

   Ghdl_Release : constant String :=
     "(2.0.0.r143.g01c330faa)" &
     " [Dunoon edition]";
end Version;
```

Which is expected. However after running ``sudo make install``, version.tmp contains

```
--  This file is automatically generated by Makefile

package Version is
   Ghdl_Ver     : constant String :=
     "3.0.0-dev";

   Ghdl_Ref     : constant String :=
     "unknown";

   Ghdl_Hash    : constant String :=
     "unknown";

   Ghdl_Release : constant String :=
     "(tarball)" &
     " [Dunoon edition]";
end Version;
```

The reason for this can be seen in the log output:

```
fatal: unsafe repository ('/home/lukas/work/repo/ghdl/ghdl' is owned by someone else)
To add an exception for this directory, call:

	git config --global --add safe.directory /home/lukas/work/repo/ghdl/ghdl
```

Which originates from the `git describe --dirty --long` call on line 218 of Makefile.in. This call works fine when running ``make`` as the normal user, but when ``sudo make install`` is run, i.e. running as root, it fails due to being "unsafe".

This new git unsafe check was introduced in version 2.35.2  of git (https://raw.githubusercontent.com/git/git/master/Documentation/RelNotes/2.35.2.txt) due to a security issue (CVE-2022-24765).  The Ubuntu team backported this patch to earlier releases as well on 8th April (http://changelogs.ubuntu.com/changelogs/pool/main/g/git/git_2.25.1-1ubuntu3.3/changelog).

Some discussion is available in this thread: https://github.com/actions/checkout/issues/760

In summary this issue affects all users that have a new git version, as well as pretty much all Ubuntu users.

# Discussion

Indeed running the ``sudo git config --global --add safe.directory /home/lukas/work/repo/ghdl/ghdl`` call recommended by the error printout solves the problem. Version.tmp is correct and ``ghdl --version`` reports 

```
GHDL 3.0.0-dev (2.0.0.r143.g01c330faa) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 mcode code generator
Written by Tristan Gingold.
```

Though perhaps there is a solution that is not manual, and hence more sustainable over time? I am not very good with Makefiles so I am afraid I can not contribute much here. But perhaps the generation of ``version.tmp`` can be done when running ``make`` but omitted when running ``sudo make install``?

tgingold, 2022-05-02:
Interesting; but I am not sure what we can do.

You can of course set the git global config, or remove the 'force' on the `version.tmp:` line.
Or we could change the `make install` rule to not depend on anything and then the user would be responsible for doing `make all` before `make install`.  But that would be unusual.

Or we could avoid to recreate `version.tmp` is the user is not the owner of the file.  But would that work on Windows ?

tgingold, 2022-05-02:
Command `test -O file` can be used to test if `file` is owned by the current user.