ghdl/ghdl #2048: INTERNAL error with ghdl --synth

mikey, 2022-05-02:
**Description**
I get an internal error with tip and this code

**Expected behaviour**
No INTERNAL ERROR

**How to reproduce?**
```
% cat test.vhdl
library ieee;
use ieee.all.common.workall;
entity icache is end;
% ghdl --synth --std=08 --no-formal  test.vhdl -e icache
sem_selected_by_all_name: cannot handle IIR_KIND_LIBRARY_DECLARATION (test.vhdl:1:9)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r143.g01c330faae) [Dunoon edition]
Compiled with GNAT Version: 11.2.1 20220127 (Red Hat 11.2.1-
Target: x86_64-redhat-linux
/home/mikey/src/microwatt2/
Command line:
ghdl --synth --std=08 --no-formal test.vhdl -e icache
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x5cc343 0x7114b7 0x714863 0x70e8ee 0x71484b 0x7159b9 0x6ea6f2 0x6eaa5b 0x6ead6c 0x6eb2d4 0x6c1e52 0x6c2cbc 0x6c2f18 0x6c5b15 0x745dde 0x7467c7 0x656e45 0x7f31fb 0x40bff5 0x7f4654ab355e 0x7f4654ab360a 0x40a5c3 0xfffffffffffffffe
******************************************************************
```

```vhd :file: test.vhdl
library ieee;
use ieee.all.common.workall;
entity icache is end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl --synth --std=08 --no-formal  test.vhdl -e icache
```

**Context**
Please, provide the following information:

- OS: Linux
- Origin:

  - [x] Built from sources: `commit 01c330faae`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r143.g01c330faae) [Dunoon edition]
Compiled with GNAT Version: 11.2.1 20220127 (Red Hat 11.2.1-
Target: x86_64-redhat-linux
/home/mikey/src/microwatt2/
Command line:
ghdl --synth --std=08 --no-formal test.vhdl -e icache
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x5cc343 0x7114b7 0x714863 0x70e8ee 0x71484b 0x7159b9 0x6ea6f2 0x6eaa5b 0x6ead6c 0x6eb2d4 0x6c1e52 0x6c2cbc 0x6c2f18 0x6c5b15 0x745dde 0x7467c7 0x656e45 0x7f31fb 0x40bff5 0x7f4654ab355e 0x7f4654ab360a 0x40a5c3 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Found using @antonblanchard [microwatt](https://github.com/antonblanchard/microwatt/) and cvise to reduce.

umarcor, 2022-05-02:
I can confirm the error with either mcode or LLVM using either VHDL 1993 or VHDL 2008.

closed 2022-05-02 by tgingold

mikey, 2022-05-02:
This fixes the original bug but I can now hit the same problem with this code 
```
library ieee;
use ieee.std_logic_1164.all.common.workall;
entity icache is end;
```
I'm generating this error by the attached test.vhdl code and running [cvise](https://github.com/marxin/cvise). Here's how I'm doing it. This might be more useful than submitting each bug I hit individually. 
```
% cat ./ghdl-cvise.sh  
#!/bin/bash

ghdl --synth --std=08 --no-formal test.vhdl -e icache 2>&1 |grep "Exception TYPES.INTERNAL_ERROR"

exit $?
% cvise --timeout 5 ./ghdl-cvise.sh  test.vhdl
00:00:00 INFO ===< 3038894 >===
00:00:00 INFO running 4 interestingness tests in parallel
00:00:00 INFO INITIAL PASSES
00:00:00 INFO ===< IncludesPass >===

.............. SNIPPED ................

Runtime: 202 seconds
Reduced test-cases:

--- /home/mikey/src/microwatt/test.vhdl ---
library ieee;
use ieee.std_logic_1164.all.common.workall;
entity icache is end;

%
```
[test.vhdl.gz](https://github.com/ghdl/ghdl/files/8606305/test.vhdl.gz)

tgingold, 2022-05-03:
Interesting.  I will try cvise.

A long time ago, I tried to find a tool to automatically generate files according to a grammar, but with errors.  I was not very successful.

mikey, 2022-05-03:
Thanks @tgingold 

Here's another one I'm having issues with. Slightly different grep in the interestingness test for cvise.

Original test case is here [test.vhdl.gz](https://github.com/ghdl/ghdl/files/8608773/test.vhdl.gz)

```
% cat ghdl-cvise.sh
#!/bin/bash

ghdl --synth --std=08 --no-formal  test.vhdl -e icache 2>&1 |grep "netlists-memories.adb:1419"
exit $?

% cvise --timeout 10 --not-c ghdl-cvise.sh test.vhdl
00:00:00 INFO ===< 1834962 >===
00:00:00 INFO running 4 interestingness tests in parallel
00:00:00 INFO INITIAL PASSES
00:00:00 INFO ===< BlankPass >===
00:00:00 INFO (0.5%, 10115 bytes, 258 lines)

.............. SNIPPED ................

Runtime: 224 seconds
Reduced test-cases:

--- /home/mikey/src/microwatt/test.vhdl ---
                 package wishbone_types is     constant wishbone_data_bits : integer := 64;
                        end ;
                        library ieee;
                        use ieee.std_logic_1164.all;
                        package common is     constant REAL_ADDR_BITS : positive := 6;
                            subtype real_addr_t is std_ulogic_vector(REAL_ADDR_BITS - 1 downto 0);
                            function addr_to_real(addr: std_ulogic_vector(63 downto 0)) return real_addr_t;
                            type Fetch1ToIcacheType is record 	req: std_ulogic;
                                virt_mode : std_ulogic;
                        	nia: std_ulogic_vector(63 downto 0);
                                big_endian: std_ulogic;
                            end record;
                            type MmuToIcacheType is record         tlbld : std_ulogic;
                                tlbie : std_ulogic;
                                pte   : std_ulogic_vector(63 downto 0);
                            end record;
                        end ;
                        package body common is     function addr_to_real(addr: std_ulogic_vector(63 downto 0)) return real_addr_t is     begin         return addr(real_addr_t'range);
    end;
end common;
library ieee;
use ieee.std_logic_1164.all;
use work.common.all;
use work.wishbone_types.all;
entity icache is
    port (
        clk          : in std_ulogic;
        rst          : in std_ulogic;
        i_in         : in Fetch1ToIcacheType;
        m_in         : in MmuToIcacheType;
	stall_out    : out std_ulogic
        );
end entity icache;
architecture rtl of icache is
    function log2(i : natural) return integer is
        variable ret : integer := 0;
    begin
        return ret;
    end function;
    constant LINE_SIZE : positive := 64;
    constant ROW_SIZE  : positive := wishbone_data_bits / 8;
    constant NUM_LINES : positive := 2;
    constant NUM_WAYS  : positive := 4;
    constant TLB_SIZE : positive := 4;
[... 57 lines cut ...]
       		end if;
       	    end if;
       	end loop;
       	stall_out <= is_hit;
           end process;
       end;

% ghdl --synth --std=08 --no-formal  test.vhdl -e icache 2>&1 |tail -15

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r151.g176b14533b) [Dunoon edition]
Compiled with GNAT Version: 11.3.1 20220421 (Red Hat 11.3.1-
Target: x86_64-redhat-linux
/home/mikey/src/microwatt2/
Command line:
ghdl --synth --std=08 --no-formal test.vhdl -e icache
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-memories.adb:1419
Call stack traceback locations:
0x54b62a 0x54b7dc 0x54b935 0x54b903 0x54f755 0x681bb6 0x6b024b 0x6b0296 0x681b2c 0x7469a2 0x656e45 0x7f3323 0x40bff5 0x7f0d964e1f1e 0x7f0d964e1fce 0x40a5c3 0xfffffffffffffffe
******************************************************************
%
```

tgingold, 2022-05-04:
This one is a different issue, as the crash appears in the memory inference.

mikey, 2022-05-04:
Do you want me to raise a new issue on github?

mikey, 2022-05-06:
@tgingold did you manage to get anywhere with this last issue?

tgingold, 2022-05-06:
Yes, it's a crash in the memory inference code.  There are already a few issues about that, so I have to dig.

tgingold, 2022-10-11:
Yes please.  But I don't plan to work on it immediately.

commit 898b6011 (2022-05-02): vhdl-sem_names(sem_selected_by_all_name): avoid a crash

Emit an error message on invalid library declaration prefix.
Fix #2048