ghdl/ghdl #2049: Synthesis crashes with STORAGE_ERROR

marzoul, 2022-05-02:
**Description**
Synthesis of a design makes GHDL crash with message : `raised STORAGE_ERROR : stack overflow or erroneous memory access`
The issue is dependent on design size : below a certain threshold it passes, above it crashes.
RAM usage is around 20 MB so this is not a limited-memory kind of issue.

The reproducer was created by reducing a much larger design to something as minimal as possible.
There is no actual functionality, just instantiation of unused things (constant array initialized by function call, external component and processes). What triggers the crash is different that just-fixed #2046 and there may be more than one cause given that either instantiation of LUT component, or having a process, leads to the crash.

Synthesis passes with generic `SIZE=2`, but fails with generic `SIZE=10`
Synthesis passes with constant `BLOCKS=1000`, but fails with generic `BLOCKS=5000`
Other tweaks in the VHDL code can make the synthesis pass ; see comments.

**Expected behaviour**
No crash, and correct synthesis results.

**How to reproduce?**

```vhd :file: engine.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity phony is
	port (
		i : in  std_logic;
		o : out std_logic
	);
end entity;

architecture synth of phony is

	constant BLOCKS : natural := 5000;
	constant BITS   : natural := 5;

	-- Unused but necessary to reproduce the crash : passing a vector is necessary
	function func1(A : std_logic_vector) return natural is
	begin
		return 1;
	end function;

	function func2(b : boolean) return natural is
	begin
		return func1(std_logic_vector(to_unsigned(0, BITS)));
	end function;

	-- Unused but necessary to reproduce the crash
	constant x : natural := func2(true);

	component LUT is
		port (
			O  : out std_logic;
			I0 : in  std_logic
		);
	end component;

begin

	gen : for b in 0 to BLOCKS-1 generate

		-- Both of these, lut and process, can reproduce the crash

		l : LUT
			port map (
				O  => open,
				I0 => '0'
			);

		p : process(all)
			variable v : natural;
		begin
			v := BITS;
		end process;

	end generate;

	o <= i;

end architecture;

library ieee;
use ieee.std_logic_1164.all;

library work;
use work.all;

entity engine is
	generic (
		SIZE : natural := 10
	);
	port (
		i : in std_logic;
		o : out std_logic
	);
end entity;

architecture synth of engine is

begin

	gen : for c in 0 to SIZE-1 generate

		ph : entity phony
			port map (
				i => '0',
				o => open
			);

	end generate;

	o <= i;

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 engine.vhd
ghdl --synth --std=08 -gSIZE=10 engine > engine_synth.vhd
```

**Context**

- OS: Archlinux
- Origin: AUR package built from sources: `commit 01c330faaed81ecd4a38bfcdc4838ea0f518e5d7`

```
commit 01c330faaed81ecd4a38bfcdc4838ea0f518e5d7 (HEAD -> makepkg, origin/master, origin/HEAD)
Author: Tristan Gingold <tgingold@free.fr>
Date:   Mon May 2 08:00:42 2022 +0200
```

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r143.g01c330faa) [Dunoon edition]
Compiled with GNAT Version: 11.2.0
Target: x86_64-pc-linux-gnu
/home/prostboa/Projets/cellver/cellver/fpga/ghdl-crash-2/
Command line:
ghdl --synth --std=08 -gSIZE=10 engine
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7f92c92ce55e 0x55821eb2117b 0x55821eb2120e 0x55821eb2b79c 0x55821eb43efc 0x55821eb4efeb 0x55821eb53882 0x55821eb542b5 0x55821eb520ae 0x55821eb53e2a 0x55821eb542b5 0x55821eb15068 0x55821eb15463 0x55821eb2d38a 0x55821eb2d5d0 0x55821eb2d3b8 0x55821eb2dd2a 0x55821ebb2027 0x55821eb11d68 0x55821ec2d7b5 0x55821e9836dd 0x7f92c92b930e 0x7f92c92b93bf 0x55821e983753 0xfffffffffffffffe
******************************************************************
```

umarcor, 2022-05-02:
Interestingly, I can reproduce the crash with `GHDL release: 3.0.0-dev (v2.0.0-143-g01c330fa)` (using container `gcr.io/hdl-containers/ghdl`) (mcode), but it is successful on MSYS2 using `GHDL 3.0.0-dev (2.0.0.r77.g4775ebbd6)` (LLVM). Therefore, it might be specific to mcode or a regression.

umarcor, 2022-05-02:
The reproducer requires VHDL 2008 because of `p : process(all)`. However, that process is "useless". If the process is removed, the reproducer can be analysed with VHDL 1993. The same crash is produced when using mcode on Linux. This time, the crash is reproducible on MSYS2 with LLVM as well, but it's a different bug report.

```
root@595b761c4487:/wrk# ghdl synth -gSIZE=10 engine
engine.vhd:45:17:warning: instance "l" of component "lut" is not bound [-Wbinding]
                l : LUT
                ^
engine.vhd:12:14:warning: (in default configuration of phony(synth))

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (v2.0.0-143-g01c330fa) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/wrk/
Command line:
ghdl synth -gSIZE=10 engine
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7fc91898213e 0x563f4f56202c 0x563f4f5620b2 0x563f4f562140 0x563f4f5520cc 0x563f4f55d3de 0x563f4f55dc97 0x563f4f55a7b6 0x563f4f55d2c8 0x563f4f536151 0x563f4f53875a 0x563f4f5389c4 0x563f4f53d354 0x563f4f55d3bc 0x563f4f53b387 0x563f4f53c910 0x563f4f538119 0x563f4f538820 0x563f4f5389c4 0x563f4f53d354 0x563f4f5046d0 0x563f4f5075bb 
0x563f4f507bf1 0x563f4f507d75 0x563f4f506a97 0x563f4f50691b 0x563f4f5069c1 0x563f4f50688a 0x563f4f5069c1 0x563f4f506ac2 0x563f4f507386 0x563f4f58c0d3 0x563f4f498513 0x563f4f63e001 0x563f4f235ca3 0x7fc9187b3d08 0x563f4f234288 0xfffffffffffffffe
******************************************************************
```

```
# ghdl --synth -gSIZE=10 engine
engine.vhd:45:17:warning: instance "l" of component "lut" is not bound [-Wbinding]
                l : LUT
                ^
engine.vhd:12:14:warning: (in default configuration of phony(synth))

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r77.g4775ebbd6) [Dunoon edition]
Compiled with GNAT Version: 11.2.0
Target: x86_64-w64-mingw32
E:\
Command line:
C:\msys64\mingw64\bin\ghdl.exe --synth -gSIZE=10 engine
Exception PROGRAM_ERROR raised
Exception information:
raised PROGRAM_ERROR : EXCEPTION_ACCESS_VIOLATION
Call stack traceback locations:
0x7ff62042009b 0x7ff8fd6720cd 0x7ff8fd621452 0x7ff8fd670bfc 0x7ff62034e3d2 0x7ff62034e44a 0x7ff62034e4b9 0x7ff620386060 0x7ff62037bb93 0x7ff62037c51e 0x7ff620378d72 0x7ff62037ba74 0x7ff62035daf3 0x7ff62035fc18 0x7ff62035fefa 0x7ff6203645dd 0x7ff62037bb6f 0x7ff6203627e6 0x7ff620363c8c 0x7ff62035f60d 0x7ff62035fd0e 0x7ff62035fefa 
0x7ff6203645dd 0x7ff62036db1b 0x7ff620338c46 0x7ff62033925e 0x7ff6203393bb 0x7ff62033a79b 0x7ff62033a683 0x7ff62033a60e 0x7ff62033a7be 0x7ff62033b0ab 0x7ff6203d4de3 0x7ff6202d904d 0x7ff6203de4b0 0x7ff620162cbe 0x7ff6201613af 0x7ff6201614e4 0x7ff8fd367032 0x7ff8fd62264f
******************************************************************
```

closed 2022-05-02 by tgingold

marzoul, 2022-05-02:
Great ! My big design now passes synthesis and is processed successfully through Yosys too \o/
Thanks !

tgingold, 2022-05-04:
Great news!

commit fd518765 (2022-05-02): synth-vhdl_context: resize table before access. Fix #2049