ghdl/ghdl #2052: GHDL crashes when using a full "hacked" 2^32 byte array.

tristanitschner, 2022-05-06:
**Description**

So recently I have been trying to work around the fact, that the maximum integer size in VHDL is 2^32 - 1. Given the amount of ram available nowadays, this shouldn't be a problem from a hardware perspective. This is what I came up with:

```vhdl
library ieee;
use ieee.numeric_std.all;
use ieee.std_logic_1164.all;

entity no_more_greater_than_pow_2_31_fuckery is 
  generic(
    width : integer := 8
         );
end;
architecture a_no_more_greater_than_pow_2_31_fuckery of no_more_greater_than_pow_2_31_fuckery is 
  subtype byte is std_logic_vector(width - 1 downto 0);
  type ram is array(integer range <>) of byte;
  type bank is array(integer range<>) of ram;

  signal mem : bank
  (2**16-1 downto 0)
  (2**16-1 downto 0);
  signal index : unsigned(32-1 downto 0) := (others => '0');

  function read(memx : bank; indexx : unsigned) return byte is
  begin
    return memx
    (to_integer(indexx(31 downto 16)))
    (to_integer(indexx(15 downto 0)));
  end function;

  procedure write(signal memx : inout bank; indexx : unsigned; value : byte) is
  begin
    memx
    (to_integer(indexx(31 downto 16)))
    (to_integer(indexx(15 downto 0))) 
    <= value;
  end procedure;

  signal clk,rst : std_logic := '1';
  constant clk_period : time := 10 ns;
begin

  clk <= not clk after clk_period/2;
  rst <= '1', '0' after clk_period*5;

  process (all) is
  begin
    if rising_edge(clk) then
      index <= index + "1";
      write(mem, index, std_logic_vector(index(byte'range)));
      if index > "0" then
        report "Wrote to memory the value " & to_hstring(read(mem, index - "1"));
      end if;
    end if;
  end process;

end;
```

Now when running
```sh
	ghdl -a --std=08 no_more_greater_than_pow_2_31_fuckery.vhd
	ghdl -e --std=08 no_more_greater_than_pow_2_31_fuckery
	ghdl -r --std=08 no_more_greater_than_pow_2_31_fuckery
```

this is what I get:

```sh
ghdl(32540,0x1182a1dc0) malloc: Incorrect checksum for freed object 0x7ff98ed37cd8: probably modified after being freed.
Corrupt value: 0x0
ghdl(32540,0x1182a1dc0) malloc: *** set a breakpoint in malloc_error_break to debug
^C^C^Z
```

Note that it was not possible to kill ghdl with ^C.

```sh
$ ghdl --version
GHDL 3.0.0-dev (2.0.0.r63.gab2e2eaa) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 mcode code generator
Written by Tristan Gingold.
```

In fact, I somewhat didn't expect this to work, but a proper error message would be very nice :)

diogratia, 2022-05-07:
You can make the test case even more compact:

```vhdl
library ieee;
use ieee.numeric_std.all;
use ieee.std_logic_1164.all;

entity no_more_greater_than_pow_2_31_fuckery is 
  generic(
    width : integer := 8
         );
end;
architecture a_no_more_greater_than_pow_2_31_fuckery of no_more_greater_than_pow_2_31_fuckery is 
  subtype byte is std_logic_vector(width - 1 downto 0);
  type ram is array(integer range <>) of byte;
  type bank is array(integer range<>) of ram;

  signal mem : bank
  (2**16-1 downto 0)
  (2**16-1 downto 0);
  -- signal index : unsigned(32-1 downto 0) := (others => '0');

  -- function read(memx : bank; indexx : unsigned) return byte is
  -- begin
  --   return memx
  --   (to_integer(indexx(31 downto 16)))
  --   (to_integer(indexx(15 downto 0)));
  -- end function;
  --
  -- procedure write(signal memx : inout bank; indexx : unsigned; value : byte) is
  -- begin
  --   memx
  --   (to_integer(indexx(31 downto 16)))
  --   (to_integer(indexx(15 downto 0)))
  --   <= value;
  -- end procedure;

  -- signal clk,rst : std_logic := '1';
  -- constant clk_period : time := 10 ns;
begin

  -- clk <= not clk after clk_period/2;
  -- rst <= '1', '0' after clk_period*5;

  -- process (all) is
  -- begin
  --   if rising_edge(clk) then
  --     index <= index + "1";
  --     write(mem, index, std_logic_vector(index(byte'range)));
  --     if index > "0" then
  --       report "Wrote to memory the value " & to_hstring(read(mem, index - "1"));
  --     end if;
  --   end if;
  -- end process;

end;
```

After that start running the multidimensional array index ranges down. 

A little light googling `malloc: Heap corruption detected, free list is damaged` finds this [plagued the heck out of the Swift guys based on boundary alignment issues](https://topologyeyewear.github.io/engineering-blog/2018/03/07/heap_corruption/). (There's a follow on for topic at [Heap Corruption Follow-up: Size, Alignment, and Crashing Collections Alignment and buffers and tails, oh my!](https://topologyeyewear.github.io/engineering-blog/2018/03/19/heap_corruption_followup/)).

The allocation problem in the Swift compiler, discovered after iOS 11 was released had to do structure alignment. It's likely the problem in GHDL is similar.

On a ghdl-macos-11.0-llvm nightly build GHDL 3.0.0-dev (2.0.0.r99.gde8bb483), reducing the power of 2 in the indices bound below 2**13-1 and the problem quit showing up _immediately_. 

I got bored watching the amount of virtual memory going up with smaller values after crossing 42GB (my Late 2020 Macbook Air M1 started to slow the VM size increase, it may clear out some of un-accessible update cache cruft from my SSD[1]). Signals are really memory intensive. It finished sometime later without reporting the error (and I didn't see the VM size). It's unclear how to determine whether a particular platform can support the full size array after any fix. 

I always thought -2019 should add a new predefined integer type instead of increasing the range of INTEGER which is used for index ranges. Really large arrays are seldom found in hardware before abstraction. Notice SDRAMs are in RAS granularity arrays or smaller and 2**31 -1 seems a bit excessive. I don't know of any end user platforms that have the memory to give any meaningful simulation performance. By the time we do we'll like move on from HDL 'centric simulators anyway.

Here the message should cause a fatal error and hopefully a bit more in the way of diagnostics, tough at run time.

<hr>

[1] It did! MacOS 12.3.1 Monterey, 8GB, 256 GB SSD. Certain categories of downloads get cached in the raw file system just in case they get shared locally. Disk storage in MacOS is complex.

dalex78, 2026-08-25:
# Issues #812, #1111 and #2052 — running out of memory is reported as a GHDL bug

Analysis by an AI agent.

## Why these three are one change

All three reports are a design that needs more memory than the machine has, and in each case GHDL does not say so. What it does instead depends on which backend is running, and both answers were wrong:

- with mcode, `Storage_Error` escapes into the internal-error box ("GHDL Bug occurred"), sending the user to the bug tracker for something no maintainer can fix;
- with gcc and llvm, the simulation executable gets a null pointer back from a failed allocation, stores through it, and dies on `SIGSEGV`, which grt reports as `NULL access dereferenced` — i.e. it blames the user's design.

The designs differ, the reason each is large differs, but the defect they expose is one and the same, so this is one fix referenced from all three issues rather than three copies of the same patch.

- **#812** — `array(0 to 2**28-1) of std_logic_vector(127 downto 0)` as a signal. tgingold worked the arithmetic out on the thread: *"A signal needs at least 144bytes, so you need at least 9GB for the signals"*, and the reporter agreed once the factor of 12 was pointed out. Nothing here is a leak; the design really is too big, and the complaint is that GHDL does not say so.
- **#1111** — a `std_logic_vector` sized from a generic close to `INTEGER'HIGH`.
- **#2052** — a nested array signal, 2**16 x 2**16 bytes. The reporter asked for exactly this: *"I somewhat didn't expect this to work, but a proper error message would be very nice"*.

## What the fix does

Three places, one message.

1. `Grt.Errors.Out_Of_Memory` (new, exported to C as `__ghdl_out_of_memory`) reports

   ```
   error: out of memory -- the design requires more memory than is available on this machine
   ```

   through the ordinary error path and exits with a failure status.

2. grt's allocation entry points call it instead of handing back a null pointer: `__gnat_malloc` and `__gnat_realloc` in `grt-cgnatrts.c` — which is what *every* `new` in grt goes through, including `Grt.Signals.Create_Signal`, the one that actually fails in #1111 and #2052 — and `Ghdl_Malloc` / `Ghdl_Malloc0` in `grt-lib.adb`, which the generated code calls directly.

3. `Storage_Error` is caught in the three places where GHDL otherwise falls into the internal-error box: the driver (`ghdlmain.adb`), the synthesis entry point (`libghdl_synth.adb`) and the translation front end (`ortho_front.adb`). This is the mcode side, where the simulation runs inside the `ghdl` process against the full Ada runtime and a failed allocation is an exception rather than a null pointer.

## What it does not do

It does not make any of these designs elaborate, and it is not meant to.

It does not address the separate point tgingold raised on #1111 — *"Length computation is done on 32 bit; but I agree that here this is a size computation"*. `Grt.Types.Ghdl_Index_Type` is still `mod 2 ** 32`, so a single object larger than 4 GB still cannot be described. The specific symptom in that report (a size of `0x80001000` sign-extended to `0xffffffff80001000` and handed to `malloc`) could not be reproduced here, and would need about 40 GB of RAM to reach at all; what is reproducible today is the failed allocation, and that is what is fixed. The 32-bit size computation is worth keeping #1111 open for, or re-scoping it to.

It also does not fix #812 on gcc and llvm, and that is a different limitation, not an oversight. There the initial value of the signal is built in a stack temporary, so the generated `DECL_ELAB` has a 32 GB stack frame — llvm even warns while analysing:

```
warning: cosim_test.vhd:6:0: stack frame size (34359738504) exceeds limit (4294967295)
   in function 'work__cosim_test__ARCH__rtl__DECL_ELAB'
```

and the executable dies on stack overflow before it allocates anything, so no allocation check can catch it. gcc prints no warning at all and simply segfaults. Making a statically huge object elaborate out of the heap is a separate change; #812 should stay open for it.

## Testing

- `testsuite/gna/issue1111` and `testsuite/gna/issue2052` analyse the reporter's design and elaborate it, expecting the out-of-memory diagnostic and asserting that no internal-error box appears. They fail on all three backends without this fix and pass with it.
- `testsuite/gna/issue812` does the same on the JIT backends (mcode and llvm-jit). On gcc and llvm it only checks that the design still analyses, because of the stack frame above; the reason is spelled out in the test.

Elaboration runs under a virtual-memory cap. Without one, GHDL allocates until the machine runs out, which is slow and hostile to whatever else is running; with a cap the failure is immediate and contained. Shells that do not support `ulimit -v` skip the check rather than run unbounded.

Where the cap goes depends on the backend, and getting this wrong is what made the first version of these tests fail in CI on the Windows and Ubuntu 26.04 runners while passing on the others. The JIT backends allocate the design in `ghdl -e` itself, so the cap has to cover that step. The gcc and llvm backends only allocate when the elaborated executable runs, and their `ghdl -e` spends its time in the code generator and the linker instead — whose memory use is a property of the toolchain installed on the machine, not of the design. Capping *those* is what broke: 2 GB was enough for the toolchain of one runner and not of another. They are now kept out of the cap, which only covers `ghdl -r`.

The three branches these came from each added a reproducer with no `testsuite.sh` at all, so the gna suite tried to execute the directory and failed on every backend. That is fixed here.