ghdl/ghdl #2053: VHDL 2008 type parameters: elaboration fails for blocks statements, synthesis fails for entities

tristanitschner, 2022-05-06:
Excited by taking note that VHDL 2008 does in fact support type parameters, and thus makes possible writing templated code similar to templates in C++ I went ahead and implemented a simple example, such that I don't have to implement a separate FIFO for every datatype in use:

```vhdl
library ieee;
use ieee.numeric_std.all;
use ieee.std_logic_1164.all;

entity generic_fifo_fwft_b is 
  generic (
            size : integer := 256;
            async_reset : boolean := false
          );
  port (
         clk     : in  std_logic;
         rst     : in  std_logic;
         datain  : in  std_logic_vector(7 downto 0);
         dataout : out std_logic_vector(7 downto 0);
         empty   : out std_logic;
         full    : out std_logic;
         wr      : in  std_logic;
         rd      : in  std_logic
       );
end;
architecture a_generic_fifo_fwft_b of generic_fifo_fwft_b is 
type mystream_t is record
  x : std_logic_vector(3 downto 0);
  y : integer range 0 to 2**4-1;
end record;
signal imin  : mystream_t;
signal imout : mystream_t;
begin

  dataout <= imin.x & std_logic_vector(to_unsigned(imin.y, 4));
  imin.x <= datain(7 downto 4);
  imin.y <= to_integer(unsigned(datain(3 downto 0)));

  fifo: block is
  generic(
           type stream_t
  );
  generic map (
                stream_t => mystream_t
              );
  port(
        bdatain  : in  stream_t;
        bdataout : out stream_t
      );
  port map (
             bdatain  => imin,
             bdataout => imout
           );

  type memory_t is array(size-1 downto 0) of stream_t;
  signal wrptr : integer range 0 to size - 1;
  signal rdptr : integer range 0 to size - 1;
  signal mem : memory_t;
  signal inverted : boolean;
  begin

    empty <= '1' when (rdptr = wrptr) and not inverted else '0';
    full  <= '1' when (rdptr = wrptr) and     inverted else '0';

    bdataout <= mem(rdptr);

    process (all) is
    begin
      if rising_edge(clk) then
        if wr and not full then
          mem(wrptr) <= bdatain;
          wrptr <= wrptr + 1;
        end if;
        if rd and not empty then
          rdptr <= rdptr + 1;
        end if;
        if wr and rd then
          null;
        elsif wr and not full then
          inverted <= not inverted when wrptr + 1 mod size < wrptr;
        elsif rd and not empty then
          inverted <= not inverted when rdptr + 1 mod size < rdptr;
        end if;
        if not async_reset then
          if rst then
            inverted <= false;
            rdptr <= 0;
            wrptr <= 0;
          end if;
        end if;
      end if;
      if async_reset then
        if rst then
          inverted <= false;
          rdptr <= 0;
          wrptr <= 0;
        end if;
      end if;
    end process;
  end block;

end;
```
(Note that this code is untested. I already spotted a bug, however for the purpose of this bug report it's fine.)

It doesn't elaborate fine:

```sh
ghdl -a --std=08 generic_fifo_fwft.vhd generic_fifo_fwft_b.vhd
ghdl -e --std=08 generic_fifo_fwft_b
translate_subtype_definition: cannot handle IIR_KIND_INTERFACE_TYPE_DEFINITION (generic_fifo_fwft_b.vhd:36:17)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r63.gab2e2eaa) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Volumes/Seagate/git/_mygit/fuckado/stream_tools/
Command line:
ghdl -e --std=08 generic_fifo_fwft_b
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Load address: 0x10a251000
Call stack traceback locations:
0x10a41e5ea 0x10a5b069e 0x10a5b0d16 0x10a61182a 0x10a618244 0x10a63cd4b 0x10a63d082 0x10a601304 0x10a5f4c75 0x10a6060c6 0x10a60034e 0x10a646653 0x10a585dd4 0x10a4b9d72 0x10a649c5c 0x10a25655c
******************************************************************
```

However, when using an entity:
```vhdl
library ieee;
use ieee.numeric_std.all;
use ieee.std_logic_1164.all;

entity generic_fifo_fwft is 
  generic (
    type stream_t;
    size : integer := 256;
    async_reset : boolean := false
  );
  port (
         clk     : in  std_logic;
         rst     : in  std_logic;
         datain  : in  stream_t;
         dataout : out stream_t;
         empty   : out std_logic;
         full    : out std_logic;
         wr      : in  std_logic;
         rd      : in  std_logic
       );
end;
architecture a_generic_fifo_fwft of generic_fifo_fwft is 
  type memory_t is array(size-1 downto 0) of stream_t;
  signal wrptr : integer range 0 to size - 1;
  signal rdptr : integer range 0 to size - 1;
  signal mem : memory_t;
  signal inverted : boolean;
begin

  empty <= '1' when (rdptr = wrptr) and not inverted else '0';
  full  <= '1' when (rdptr = wrptr) and     inverted else '0';

  dataout <= mem(rdptr);

  process (all) is
  begin
    if rising_edge(clk) then
      if wr and not full then
        mem(wrptr) <= datain;
        wrptr <= wrptr + 1;
      end if;
      if rd and not empty then
        rdptr <= rdptr + 1;
      end if;
      if wr and rd then
        null;
      elsif wr and not full then
        inverted <= not inverted when wrptr + 1 mod size < wrptr;
      elsif rd and not empty then
        inverted <= not inverted when rdptr + 1 mod size < rdptr;
      end if;
      if not async_reset then
        if rst then
          inverted <= false;
          rdptr <= 0;
          wrptr <= 0;
        end if;
      end if;
    end if;
    if async_reset then
      if rst then
        inverted <= false;
        rdptr <= 0;
        wrptr <= 0;
      end if;
    end if;
  end process;
end;
```
and instantiate it
```vhdl
library ieee;
use ieee.numeric_std.all;
use ieee.std_logic_1164.all;

--use work.utils.all;

entity generic_fifo_fwft_inst is 
  port (
         clk     : in  std_logic;
         rst     : in  std_logic;
         datain  : in  std_logic_vector(7 downto 0);
         dataout : out std_logic_vector(7 downto 0);
         empty   : out std_logic;
         full    : out std_logic;
         wr      : in  std_logic;
         rd      : in  std_logic
       );
end;

architecture a_generic_fifo_fwft_inst of generic_fifo_fwft_inst is 
  type mystream_t is record
    x : std_logic_vector(3 downto 0);
    y : integer range 0 to 2**4-1;
  end record;
  signal imin  : mystream_t;
  signal imout : mystream_t;
begin

  dataout <= imin.x & std_logic_vector(to_unsigned(imin.y, 4));
  imin.x <= datain(7 downto 4);
  imin.y <= to_integer(unsigned(datain(3 downto 0)));

fifo: entity work.generic_fifo_fwft 
  generic map (
    stream_t    => mystream_t,
    size        => 256,
    async_reset => false
  )
  port map (
         clk     => clk,
         rst     => rst,
         datain  => imin,
         dataout => imout,
         empty   => empty,
         full    => full,
         wr      => wr,
         rd      => rd
       );
end architecture;
```

elaboration works fine. However, synthesizing does not:

```sh
ghdl -i --std=08 generic_fifo_fwft.vhd generic_fifo_fwft_inst.vhd
ghdl -m --std=08 generic_fifo_fwft_inst
ghdl --synth --std=08 --out=vhdl generic_fifo_fwft_inst > generic_fifo_fwft_inst93.vhd

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r63.gab2e2eaa) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Volumes/Seagate/git/_mygit/fuckado/stream_tools/
Command line:
ghdl --synth --std=08 --out=vhdl generic_fifo_fwft_inst
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Elements_Declaration_List
Load address: 0x103fe7000
Call stack traceback locations:
0x1043f6e51 0x1041868a7 0x104306986 0x104307199 0x1043049c6 0x104304dfc 0x104304e88 0x1043077b9 0x104309411 0x10430956a 0x1042e2b78 0x1042e2caf 0x10430a9c5 0x10430aab6 0x104303d73 0x10432f0d1 0x10424fd72 0x1043dfc5c 0x103fec55c
******************************************************************
```

I may ask: what is an Elements_Declaration_List? I couldn't find that in the LRM.

umarcor, 2022-05-07:
Thanks for providing reproducers!

This is a known issue in the sense that generic types are not fully supported yet. VHDL 2008 generics were first added to packages, then to entities (precisely for the generic FIFO use case), but it's still missing for blocks. As you found, some use cases are supported for simulation but not for synthesis yet.

Maybe you want to split this into two issues: one for tracking generic types in blocks and another one for tracking generic types in entities for synthesis.

closed 2022-05-07 by tgingold

tristanitschner, 2022-05-07:
I just noticed that type parameters still do not work in simulation after the fix, so this should probably be opened again. Or I might open another issue for it, if you prefer. Will post an update tomorrow or so.

tgingold, 2022-05-08:
type parameters are not yet supported in simulation.

commit 79a93b9f (2022-05-07): synth-vhdl_insts: handle interfaces of type interface type.  Fix #2053