ghdl/ghdl #2055: GHDL crashes when trying to implement a shift register using an array type.

erick166, 2022-05-11:
**How to reproduce?**
```
architecture func of tb is

  type   t_slv_array is array (natural range <>) of std_logic_vector;
  signal slv_array : t_slv_array(0 to 1)(9 downto 0);
  signal slv       : std_logic_vector(9 downto 0);

begin

  p_main : process
  begin

    slv_array <= slv_array(1) & slv;

    std.env.stop;
    wait;  -- to stop completely
  end process p_main;
end architecture func;
```

**Error message:**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0-dev (2.0.0.r159.g98bc5e03) [Dunoon edition]
Compiled with GNAT Version: 11.3.0
Target: i686-w64-mingw32

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:1473
Load address: 0x290000
Call stack traceback locations:
0x5a41ce 0x5ac5d5 0x5ac23e 0x5f0571 0x5f0656 0x5f0ca2 0x5f14fe 0x5f2c36 0x5f2d83 0x5f5ead 0x5fafcf 0x5c04d2 0x5b5f50 0x5c4cbb 0x5bf107 0x602b49 0x54e863 0x48b6c5 0x60581a 0x292db6 0x291384 0x7651fa27 0x773f7a7c 0x773f7a4c
******************************************************************
```

tgingold, 2022-05-12:
Thank you for the small testcase.  It's an issue due to concatenation of unbounded arrays.
I will implement it.

bradley-harden-apl, 2022-08-09:
@tgingold, would you expect to see this crash only in `llvm` and `gcc` backends but not `mcode`? That's what I'm seeing. I wonder if the `mcode` version somehow makes it through analysis when it shouldn't, which is causing the errors in #2154. Does that make any sense?

bradleyharden, 2022-08-09:
@tgingold, would you expect to see this crash only in `llvm` and `gcc` backends but not `mcode`? That's what I'm seeing. I wonder if the `mcode` version somehow makes it through analysis when it shouldn't, which is causing the errors in #2154. Does that make any sense?

tgingold, 2022-08-09:
Yes, this is expected.  mcode generated code only at elaboration.

bradleyharden, 2022-08-09:
Ah ok, then #2154 is likely an effective duplicate of this issue?

commit 14fd4bec (2022-08-11): trans-chap7: handle concat of unbounded elements.  Fix #2055
closed 2022-08-11 by tgingold