ghdl/ghdl #2062: ieee/fixed_pkg_c.vhd:8323:14:error: unhandled function: IIR_PREDEFINED_IEEE_1164_TO_STDULOGICVECTOR_SLV

donnie-j, 2022-05-17:
**Description**
Trying to synthesize RTL which uses IEEE fixed_pkg_c.vhd fails with a missing handler.

**Expected behaviour**
Synthesis succeeds

**How to reproduce?**
Apologies, I don't have a small reproducer since this error is caused by the IEEE library.  However:

foo.vhd
```
jeff@build180:~/work$ cat foo.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.fixed_pkg.all;

entity fxt is port (
   a : in  std_logic_vector(6 downto 0);
   y : out ufixed(3 downto -2));
end entity;

architecture beh of fxt is
begin
   y <= to_ufixed(a, 5, 1);
end beh;
```

Analyse, and synth
```
jeff@build180:~/work$ ghdl -a ieee/fixed_float_types_c.vhd
jeff@build180:~/work$ ghdl -a ieee/fixed_pkg_c.vhd
ieee/fixed_pkg_c.vhd:2178:25:warning: declaration of "remainder" hides function "remainder" [-Whide]
                        remainder      : UNRESOLVED_ufixed;
                        ^
ieee/fixed_pkg_c.vhd:2210:25:warning: declaration of "remainder" hides function "remainder" [-Whide]
                        remainder      : UNRESOLVED_sfixed;
                        ^
jeff@build180:~/work$ ghdl -a foo.vhd
jeff@build180:~/work$ ghdl --synth fxt
ieee/fixed_pkg_c.vhd:8323:14:error: unhandled function: IIR_PREDEFINED_IEEE_1164_TO_STDULOGICVECTOR_SLV
      arg => to_stdulogicvector (arg),
             ^
foo.vhd:9:4:warning: no assignment for port "y"
   y : out ufixed(3 downto -2));
   ^
jeff@build180:~/work$
```

I can provide copies of the IEEE fixed point libraries for testing upon request, not attaching our of an abundance of caution given IEEE's stupid, ambiguous copyright.  I've not tried the VHDL-08 versions bundled with GHDL.

tgingold, 2022-05-17:
I suppose you are using a non-standard version of fixed_pkg.
The standard version appeared with vhdl 2008 and doesn't use to_stdulogicvector (because std_ulogic_vector and std_logic_vector are now the same type).

Fix in progress.

donnie-j, 2022-05-17:
We've not yet made the transition to VHDL 08 in our DSP code.  So we're carrying around the IIRC VHDL 93 compatibility version, which is probably why it's still using std_ulogic_vector.  I think the only change we made was a minor correction to one of operator overload functions, which IIRC has long been merged upstream...

umarcor, 2022-05-17:
@donnie-j are you using the `*_c.vhdl` sources from [FPHDL/fphdl](https://github.com/FPHDL/fphdl) or some other variant?

donnie-j, 2022-05-17:
From the same lineage, but the version I have is a little different.  We have these changes (aside from setting the rounding mode, etc).  These changes appeared in 2014, and the changelog says sync with external sources, hmmm....  Might have been in response to limitations of the synthesis tool used by one of our vendors at the time, I don't remember.

```
diff -r c10e2dbefa8c -r df5713d4e39a ieee/fixed_pkg_c.vhd
--- a/ieee/fixed_pkg_c.vhd      Mon Nov 24 21:32:45 2014 -0500
+++ b/ieee/fixed_pkg_c.vhd      Sat Nov 29 15:13:14 2014 +0900
@@ -2272,24 +2272,26 @@
   function to_sulv (
     arg : UNRESOLVED_ufixed)            -- fixed point vector
     return STD_ULOGIC_VECTOR is
-    variable result : STD_ULOGIC_VECTOR (arg'length-1 downto 0);
+    subtype result_subtype is STD_ULOGIC_VECTOR (arg'length-1 downto 0);
+    variable result : result_subtype;
   begin
     if arg'length < 1 then
       return NSLV;
     end if;
-    result := STD_ULOGIC_VECTOR (arg);
+    result := result_subtype(arg);
     return result;
   end function to_sulv;

   function to_sulv (
     arg : UNRESOLVED_sfixed)            -- fixed point vector
     return STD_ULOGIC_VECTOR is
-    variable result : STD_ULOGIC_VECTOR (arg'length-1 downto 0);
+    subtype result_subtype is STD_ULOGIC_VECTOR (arg'length-1 downto 0);
+    variable result : result_subtype;
   begin
     if arg'length < 1 then
       return NSLV;
     end if;
-    result := STD_ULOGIC_VECTOR (arg);
+    result := result_subtype(arg);
     return result;
   end function to_sulv;

```

closed 2022-05-17 by tgingold