ghdl/ghdl #2065: GHDL: System assert failure

gncemre23, 2022-05-21:
**Description**
ghdl analyzing is ended up with the System assertion failure.

**Expected behaviour**
It is expecting to be compiled successfully, but it gives the system assertion error.

**How to reproduce?**

```
-- dual_port_ram.vhd
-- This file is part of bladeRF-wiphy.
--
-- Copyright (C) 2021 Nuand, LLC.
--
-- This program is free software; you can redistribute it and/or modify
-- it under the terms of the GNU General Public License as published by
-- the Free Software Foundation; either version 2 of the License, or
-- (at your option) any later version.
--
-- This program is distributed in the hope that it will be useful,
-- but WITHOUT ANY WARRANTY; without even the implied warranty of
-- MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
-- GNU General Public License for more details.
--
-- You should have received a copy of the GNU General Public License along
-- with this program; if not, write to the Free Software Foundation, Inc.,
-- 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA.

library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

entity dual_port_ram is
   generic(
      ADDR_BITS  :  in  natural := 6;
      DATA_BITS  :  in  natural := 32
   );
   port(
      clock      :  in  std_logic;
      reset      :  in  std_logic;

      acc        :  in  std_logic;
      solo       :  in  std_logic;
      write      :  in  std_logic;

      addr_a     :  in  std_logic_vector(ADDR_BITS-1 downto 0);
      in_a       :  in  std_logic_vector(DATA_BITS-1 downto 0);
      data_a     :  out std_logic_vector(DATA_BITS-1 downto 0);

      addr_b     :  in  std_logic_vector(ADDR_BITS-1 downto 0);
      in_b       :  in  std_logic_vector(DATA_BITS-1 downto 0);
      data_b     :  out std_logic_vector(DATA_BITS-1 downto 0)
   );
end entity;

architecture arch of dual_port_ram is
   type ram_t is array(natural range <>) of std_logic_vector(DATA_BITS-1 downto 0);

   signal ram : ram_t((2**ADDR_BITS-1) downto 0);
begin
   sync : process(clock, reset)
      variable add_a, add_b : integer;
   begin
      if (reset = '1') then
         for i in ram'range loop
            ram(i) <= ( others => '0' );
         end loop;
      elsif (rising_edge(clock)) then
         if (acc = '1') then
            add_a := to_integer(unsigned(addr_a));
[... 668 lines cut ...]
Please, provide the following information:

- OS: Ubuntu
- Origin:
  - [ ] Package manager: `20.04.4 LTS`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues

GHDL release: 3.0.0-dev (v2.0.0-159-g98bc5e03) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/wiphy-build/bladeRF/hdl/fpga/platforms/bladerf-micro/GHDL/build/
Command line:
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/9.1.0/ghdl1 --ghdl-fsynopsys --ghdl--std=08 --ghdl--work=wlan -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -quiet -o fft.s /home/wiphy-build/bladeRF-wiphy/fpga/ip/nuand/fft/vhdl/fft.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans.adb:434
Call stack traceback locations:
0x7f145dcfdab2 0x7c78ff 0x81ced3 0x80d09d 0x80d982 0x80d2b8 0x7f53a2 0x7d3043 0x7c4d9b 0x7c2648 0x673668 0xc1d8c2 0x669372 0x67879d 0x835649 0x66725d 0x7f145d7d5d08 0x66e0c8 0xfffffffffffffffe
******************************************************************
```

gncemre23, 2022-05-21:
I solved the issue by  separating the architectures in fft.vhd

Paebbels, 2022-05-21:
@gncemre23 Your dual port RAM wont synthesize as a RAM, because it has a reset.

-------------
@tgingold Do you want to reopen this to keep it as a testcase when you go to fix the issue later?

tgingold, 2022-05-22:
Yes, this is a simulation issue.

gncemre23, 2022-05-22:
> @gncemre23 Your dual port RAM wont synthesize as a RAM, because it has a reset.
> 
> @tgingold Do you want to reopen this to keep it as a testcase when you go to fix the issue later?
@Paebbels , if you see the architecture named synth, it does not have the reset. Synth architecture is used in fft.

Paebbels, 2022-05-22:
@gncemre23 Then your synthesis model is different from the simulation model, which is also a mistake. 

You'll simulate different behavior then it exists in hardware. E.g. your synthesis model has read-before-write behavior and your simulation model implements write-before-read. This is a completely different timing behavior in addition to the reset behavior.

gncemre23, 2022-05-23:
> @gncemre23 Then your synthesis model is different from the simulation model, which is also a mistake.
>
> You'll simulate different behavior then it exists in hardware. E.g. your synthesis model has [...]

@Paebbels I accept that there is confusion but we are using the architecture named synth one for both simulation and synthesis purposes. Actually, the architecture with reset is not instantiated anywhere. I guess it might be left from previous works. Thanks for the recommendations, anyway.

commit e260d4cf (2022-06-16): vhdl/translate: handle inertial association in recursive instantiation

Fix #2065
closed 2022-06-16 by tgingold